RobotLocomotion/drake #10225: Mobilizer::set_zero_configuration confuses state and configuration

RussTedrake, 2018-12-14:
The configuration of a mobilizer is only it's positions.
The state of a mobilizer is it's position + velocity.

The naming (or implementation) of this method is wrong:
```
  /// Sets the state stored in `context` to a _zero configuration_ as defined by
  /// set_zero_state().
  /// See set_zero_state() for details.
  void set_zero_configuration(systems::Context<T>* context) const {
    set_zero_state(*context, &context->get_mutable_state());
  }
```

amcastro-tri, 2018-12-14:
> The configuration of a mobilizer is only it's positions.
> The state of a mobilizer is it's position + velocity.

I think I really like that distinction. I think the reason I merged them together is so that there'd be a single call to initialize the state for a mobilizer.
So your proposal would be:
1) `set_zero_configuration()`: only sets qs. Useful? even if leaving vs uninitialized?
2) `set_zero_state()`, current implementation.

amcastro-tri, 2018-12-14:
my questions were answered during the review of #10226. Russ proposes to **only** have `set_zero_state()` since the name `set_zero_configuration()` is misleading.

PR #10226 (2018-12-15): deprecate Mobilizer::set_zero_configuration
resolves #10225

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10226)
commit 22b999a3 (2018-12-15): resolves #10225 (#10226)
closed 2018-12-15 by RussTedrake