RobotLocomotion/drake #10243: MultibodyPlant::GetFrameByName segfaults when invalid frame name matches model instance name

rcory, 2018-12-14:
Take the following example, where a new model is added to the `MultibodyPlant` and subsequently queried with an _invalid frame name_ (which happens to match the name of the newly added model instance):

```
const multibody::ModelInstanceIndex new_model = 
    parser.AddModelFromFile(sdf_path, "foo");
const auto& child_frame = plant_->GetFrameByName("foo", new_model);
```
This code produces a segfault. Given that the provided name is invalid, this should instead produce a run-time error alerting the user that the provided frame name is invalid.

tarquasso, 2020-05-07:
👍

amcastro-tri, 2020-05-07:
I just tried this. I do however get an exception with the message: 
> "There is no frame named 'foo' in model instance 'foo'."

Could you provide a code snippet and SDF file to reproduce this?
or even better @rcory and @tarquasso, maybe one of you could get a back-trace and locate the origin? gdb is great at that.

jwnimmer-tri, 2024-03-20:
Closing for lack of repro.

closed 2024-03-20 by jwnimmer-tri