RobotLocomotion/drake #10244: Test related to periodic publish in python is inexplicably failing

edrumwri, 2018-12-15:
This issue arose out of a Buildcop failure (see PRs #9766 and #10238). 

The CI test in question is in custom_test.py: `test_leaf_system_overrides`. This test needs some repairs irrespective of this issue because it uses floating point numbers without exact binary representations (i.e., 0.1), but relies upon the number of publishes, updates, etc. to be constant (it's hard to guarantee that in this case).  It is going to be necessary to spend some time debugging through python to discern exactly why this test is failing, however, and to correctly repair it. Numerous C++ tests of like code have been passing, and this might be a python specific problem.

sherm1, 2018-12-15:
No worries, @EricCousineau-TRI. Although this test is slightly touchy, we now think the root cause is more subtle. Drake's behavior with dt=0.1 is just like Simulink's a perfectly predictable. Don't fix anything yet please -- we'd like to make sure we can make this work as expected.

edrumwri, 2018-12-15:
> Drake's behavior with dt=0.1 is just like Simulink's and perfectly predictable

Not true, at least with what I saw yesterday between Mac OS X and Linux. But this observation was through Python, which can affect the results, as I will demonstrate in my next update to this issue.

edrumwri, 2018-12-15:
(Documenting a clue here)

I can write the same simple system in Python and in C++ that yields two different behaviors:
```
class DemoSystem(LeafSystem):
  def __init__(self):
    LeafSystem.__init__(self)
    self._DeclarePeriodicDiscreteUpdate(period_sec=1, offset_sec=0)
    self.num_updates = 0

  def _DoCalcDiscreteVariableUpdates(self, context, events, discrete_state):
    # Call base method to ensure we do not get recursion.
    LeafSystem._DoCalcDiscreteVariableUpdates(
      self, context, events, discrete_state)
    self.num_updates += 1

def main():
  system = DemoSystem() 
  simulator = Simulator(system)
  simulator.StepTo(0)

  # Should just be one update, at the beginning of StepTo().
  assert system.num_updates == 1 
```

And the C++ version:
```
class DemoSystem : public drake::systems::LeafSystem<double> {
 public:
  DemoSystem() {
    this->DeclarePeriodicDiscreteUpdate(1.0 /* period */, 0.0 /* offset */);
  }
};

GTEST_TEST(SimulatorTest, DemoSystemTest) {
  DemoSystem sys;
  Simulator<double> simulator(sys);
  simulator.StepTo(0);

  // Should just be one update, at the beginning of StepTo().
  EXPECT_EQ(simulator.get_num_discrete_updates(), 1);
}
```

The C++ test passes. The python one does not, on Linux (based on my investigations yesterday, I believe this test _will_ pass on OS X).

sherm1, 2018-12-15:
BTW our current hypothesis (admittedly far-fetched) is that the Python interpreter treats denormalized numbers differently than C++, at least on Linux. There is a floating point mode "treat denormalized numbers as zero". We think Python sets it. Does anyone know if that's true?

The cure, obviously, is for us not to depend on treatment of denormalized numbers! For all you `std::nexttoward()` fans, be aware that it can produce them :(

EricCousineau-TRI, 2018-12-15:
Yup, there's the same behavior between C++ and Python:
https://github.com/EricCousineau-TRI/drake/commit/943b1421567f22f325f8009ec17159ba1e3258f2

sherm1, 2018-12-16:
I _think_ the difference occurs at t=0. That is the only place where the denormalized number crops up. If that hypothesis is correct then it predicts that in Python there will _not_ be an update at 0 while in C++ there will be one. Neither platform should produce at update at 1 if you StepTo(1) (although one is pending at that point) because updates occur at the _beginning_ of a step.

Is that consistent with what you're seeing @EricCousineau-TRI ?

sherm1, 2018-12-16:
BTW if you change the updates to publishes, you should get one at 0 (from initialization) and one at 1 because publishes occur at the _end_ of a step.

edrumwri, 2018-12-16:
Note that I used StepTo(0) because it shows the disparity in behavior irrespective of time advancing.

EricCousineau-TRI, 2018-12-16:
Also, from a very brief review, it looks like Python 2.7 on my system (with NumPy 1.11.0) seem to represent denormalized numbers without rounding away?
StackOverflow posts [1](https://stackoverflow.com/questions/38477908/smallest-positive-float64-number) and [2](https://stackoverflow.com/questions/6063755/increment-a-python-floating-point-value-by-the-smallest-possible-amount).

On my system:
```
In [1]: import numpy as np

In [2]: x = np.finfo(float).tiny; print(x)
2.22507385851e-308

In [3]: x / 2**10
Out[3]: 2.1729236899484389e-311

In [4]: np.nextafter(0., 1.)
Out[4]: 4.9406564584124654e-324
```

edrumwri, 2018-12-16:
@EricCousineau-TRI the C++ test you provided (and marked as failing) passes just fine on my system. I take it you're running on Linux?

EricCousineau-TRI, 2018-12-16:
Yup, running Ubuntu 16.04 on my desktop machine. Can try on my laptop as well, as well as Mac.

EricCousineau-TRI, 2018-12-16:
But my biggest question is, why does `StepTo(1.)` fail on my machine? (And presumably others?)

edrumwri, 2018-12-16:
Could I get you to try something? If you replace `context_->set_time(nexttoward(current_time, -inf))` with `context_->set_time(current_time - 1e-14)` in simulator.h:678, do both tests pass?

EricCousineau-TRI, 2018-12-16:
Yup! That fixed it.

EricCousineau-TRI, 2018-12-16:
And sorry for being slow, I see now that, for whatever reason, `nexttoward` was insufficient to permit `CalcNextUpdateTime` to detect the discrete update at `t=0`, as y'all had mentioned. However, this is reproducible in C++ on my system, so it's still unclear why my system is suffering. Will see if I can capture this in a more minimal example (just comparing two numbers).

edrumwri, 2018-12-16:
That's what I expected. The denormalized number is getting used in arithmetic in leaf_system.h (GetNextSampleTime(.)), and floating point arithmetic between normalized and denormalized numbers should be avoided. I think the solution is to pick the next _normalized_ number from zero toward negative infinity. I'll ask you to try my solution before I try to take it live.

edrumwri, 2018-12-16:
> And sorry for being slow, I see now that, for whatever reason, `nexttoward` was insufficient to permit `CalcNextUpdateTime` to detect the discrete update at `t=0`, as y'all had mentioned. However, this is reproducible in C++ on my system, so it's still unclear why my system is suffering. 

Floating point math is dark magic.

EricCousineau-TRI, 2018-12-16:
Aye aye. One minor concern that I saw was the mixture of `nexttoward` (instead of `nextafter`) and `long double` (instead of `double`). Briefly checking this.

EricCousineau-TRI, 2018-12-16:
Oh jeepers... Tried doing a min repro, and this is the diff that I use that either fixes the problem or reproduces it:
```patch
diff --git a/bindings/pydrake/systems/BUILD.bazel b/bindings/pydrake/systems/BUILD.bazel
index 534ae47..f839ced 100644
--- a/bindings/pydrake/systems/BUILD.bazel
+++ b/bindings/pydrake/systems/BUILD.bazel
@@ -460,8 +460,7 @@
     name = "issue_10255_repro",
     srcs = ["issue_10255_repro.cc"],
     deps = [
-        "//systems/analysis",
-        "//systems/framework",
+        "//:drake_shared_library",
     ],
 )
```

So it *is* actually specific to Python *and* anything else that uses `libdrake.so`, so people using CMake (e.g. the underactuated course).

edrumwri, 2018-12-16:
I think it's safe to say that the problem is related to using denormalized floating point numbers and that we can avoid using those with some floating point gymnastics. I expect to have a fix tomorrow.

EricCousineau-TRI, 2018-12-16:
Yup, sounds good. I still want to know exactly why linking modality changes this behavior, but I'll investigate that separately.

As an aside, it's because something in `GetNextSampleTime` has markedly different between static linking and shared library linking:
https://github.com/EricCousineau-TRI/drake/commit/f2027d66cd7dd7ab599d4951c445fef5396829a3#diff-dfc5081278527b906fd0f3245d73dfd4R23

edrumwri, 2018-12-16:
Right. We just need an algorithm for producing next normalized floating point number _from any value_ toward -inf. I'm in the middle of testing one as we speak.

sherm1, 2018-12-16:
Doesn't `nexttoward()` work for that everywhere except 0?

edrumwri, 2019-01-03:
Addressed with PR #10260.

closed 2019-01-03 by edrumwri