RobotLocomotion/drake #10253: shared_library: Different behavior with subnormal numbers between static and shared linking

EricCousineau-TRI, 2018-12-16:
Relates #10244 - the solution there is to avoid using subnormal numbers, but I still want to understand why we have a difference in computations due to linking modality.

From that discussion:
> [...] it's because something in GetNextSampleTime has markedly different between static linking and shared library linking:
> EricCousineau-TRI@f2027d6#diff-dfc5081278527b906fd0f3245d73dfd4R23

\cc @edrumwri @sherm1

EricCousineau-TRI, 2018-12-16:
Narrowed it down to the linking of `libospray_common.so` that causes this difference:
https://github.com/EricCousineau-TRI/drake/commit/02cf96a1a97046a495bd88dd4c645f326c0efeb8#diff-ee1595d89a56753d136ae1b9683363a1R6

EricCousineau-TRI, 2018-12-16:
To follow-up on the ideal outcome, I'd settle with being able to (a) better capture this behavior (with a simple repro script) and (b) pinpoint *exactly* what line of code dependencies cause this issue.

EricCousineau-TRI, 2018-12-16:
Yup, the culprit is the fact that `ospray_common` is built with `-ffast-math`; linking `-ffast-math` indicates this behavior in general:
https://github.com/RobotLocomotion/drake/commit/10d2e772b2c6edbe09b2ce8cc38a44704c4d7d35

This satisfies (a) and (b), since we have the min-repro.

If we wanted to fix it, we should incorporate `-ffast-math` to always ensure that our tests are robust against it.

edrumwri, 2018-12-16:
Good sleuthing Eric!

sherm1, 2018-12-16:
We shouldn't be using -ffast-math (=unsafe non-IEEE math) in scientific software. It could be reasonable for gaming. Here's a random [horror story](http://pointclouds.org/news/2011/08/29/ffast-math/) from PCL. TL;DR at the bottom: "DO NOT USE -ffast-math".

sherm1, 2018-12-16:
IIRC from my game engine days, -ffast-math is necessary in order to get full performance from single precision (`float`) math but doesn't help much at all for `double` math.

EricCousineau-TRI, 2018-12-16:
> We shouldn't be using -ffast-math (=unsafe non-IEEE math) in scientific software. [...]

Agreed for our application code, though as a library Drake may be unable to assert that (e.g. if someone decides to build `bullet` with `-ffast-math`, then they decide to load Drake into the same binary - for comparison - where `set_fast_math()` has already been called).

Anywho, here's the exact offending line in OSPRay, which only happens with GCC compilation (seems like Clang does not compile it in?):
https://github.com/RobotLocomotion/drake/issues/10253#issuecomment-447676055

> Here's a random horror story from PCL. TL;DR at the bottom: "DO NOT USE -ffast-math".

Haha, that's about on par with what my debugging experience was, but thankfully already isolated to a smaller set of checks.

sherm1, 2018-12-16:
If I'm reading [this](https://lists.fedoraproject.org/pipermail/devel/2015-April/210259.html) right, the call to `set_fast_math()` done when a "bad" shared object is loaded comes from _linking_ with `-ffast-math` -- a library that was compiled with `-ffast-math` could still get some gain from instruction choice, while not mucking up the fp modes for everyone else.

jamiesnape, 2019-02-26:
Did #10262 effectively fix this?

sherm1, 2019-02-26:
Yes, between #10260 and #10262 I think this has been fully resolved. Any objections to closing it?
@EricCousineau-TRI ?

EricCousineau-TRI, 2019-02-26:
Yes; however, it'd be nice to have a small test that links against `//:drake_shared_library` and checks sub-normal numbers. Will PR a small test.

PR #10764 (2019-02-26): libdrake: Add test to ensure `-ffast-math` is not present
Resolves #10253

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10764)
closed 2019-02-26 by jamiesnape