RobotLocomotion/drake #10266: cpplint: Should use more up-to-date fork, not from styleguide

EricCousineau-TRI, 2018-12-18:
Per this issue: https://github.com/google/styleguide/pull/408

> Consider the fork at https://github.com/cpplint/cpplint (also published to Pypi) if you feel like contributing.

\cc @jwnimmer-tri @jamiesnape

EricCousineau-TRI, 2018-12-18:
Yes, it may have additional overhead of configurations changing, but it may have improvements codewise that we may want at some point.

I'd call this closeable if we at least put a comment at the top of our forked cpplint pointing to the new stuff.

jamiesnape, 2018-12-18:
> I'd call this closeable if we at least put a comment at the top of our forked cpplint pointing to the new stuff.

Sounds a little strange. I don't we should be deviating too much anyway unless the Google version is broken, whether to not there are improvements.

EricCousineau-TRI, 2018-12-18:
Aye. Will close then.

closed 2018-12-18 by EricCousineau-TRI