RobotLocomotion/drake #10270: ComputeSignedDistancePairwiseClosestPoints should compute sphere-to-sphere distance in the closed form

hongkai-dai, 2018-12-18:
Currently when we compute the signed distance between spherical objects in `ComputeSignedDistancePairwiseClosestPoints()` https://github.com/RobotLocomotion/drake/blob/master/geometry/query_object.h#L184~L185, the code resorts to FCL's GJK/EPA implementation. It approximates the spheres as polytopes, and computes the signed distance using an iterative algorithm. This is both inefficient and inaccurate. Instead we should be able to compute the signed distance between spheres in the closed form.

There are two solutions to this problem
1. Inside `ComputeSignedDistancePairwiseClosestPoints()`, it first detects if the objects are spheres. If yes, it calls a special routine to compute the distance in the closed form.
2. Fix the problem in FCL's upstream. In the signed distance query in FCL, it should have a special routine to compute the signed distance between spheres.

Solution 1 is probably a quicker fix.

SeanCurtis-TRI, 2019-01-02:
I've got a live FCL commit that will allow specific primitive tests for the *signed* distance queries. I'll submit it to FCL in the next little bit (right after I dig out of my holiday hole).

DamrongGuoy, 2019-01-02:
Thank you, @SeanCurtis-TRI.  Please keep us updated on FCL.  I update that a quick fix in Drake by PR #10281 will be ready for review when PR #10148 landed.  Happy New Year!

DamrongGuoy, 2019-01-15:
Fixed by #10281.

closed 2019-01-15 by DamrongGuoy