RobotLocomotion/drake #10276: IPOPT-Based Tests are Sometimes Flaky

liangfok, 2018-12-19:
The following errors were observed in `mac-high-sierra-unprovisioned-clang-bazel-nightly-release` CI:

```
attic/multibody/test/test_ik_pointwise.cc:79: Failure
Value of: CompareMatrices(com, expected_final, 1e-6, MatrixCompareType::absolute)
  Actual: false (Values at (2, 0) exceed tolerance: 0.90590452998206261 vs. 0.90000000000000002, diff = 0.0059045299820625896, tolerance = 9.9999999999999995e-07
m1 =
 2.6913393049638019e-12
-5.5658619815208748e-13
    0.90590452998206261
m2 =
                  0
                  0
0.90000000000000002
delta=
 2.6913393049638019e-12
-5.5658619815208748e-13
  0.0059045299820625896)
Expected: true
 1 FAILED TEST

//attic/multibody:test_ik_pointwise                                      FAILED in 0.8s
```
https://drake-cdash.csail.mit.edu/testDetails.php?test=24827506&build=1092945

```
Restoration phase is called at point that is almost feasible,
  with constraint violation 9.324139e-17. Abort.
examples/valkyrie/test/valkyrie_ik_test.cc:258: Failure
Expected equality of these values:
  info
    Which is: 100
  1
[  FAILED  ] ValkyrieIK_Test.ValkyrieIK_Test_StandingPose_Test (678 ms)
 1 FAILED TEST

//examples/valkyrie:valkyrie_ik_test                                     FAILED in 0.9s
```
https://drake-cdash.csail.mit.edu/testDetails.php?test=24827503&build=1092945

The following error was observed in `mac-high-sierra-clang-bazel-nightly-debug`:

```
Restoration phase is called at point that is almost feasible,
  with constraint violation 6.940651e-14. Abort.
Failed to solve optimization for the swing-up trajectory

//examples/pendulum:trajectory_optimization_simulation_test              FAILED in 1.5s
```
https://drake-cdash.csail.mit.edu/testDetails.php?test=24828333&build=1093112

ggould-tri, 2018-12-19:
The same error in a different test this morning:

~~~
Restoration phase is called at point that is almost feasible,
  with constraint violation 2.250082e-13. Abort.
No solution found.

//examples/acrobot:run_swing_up_traj_optimization_test                   FAILED in 1.1s
~~~

https://drake-cdash.csail.mit.edu/testDetails.php?test=24841142&build=1093816

jwnimmer-tri, 2018-12-19:
PR #10066 merged about 8 days ago, directly related to IPOPT, so it could be related.

ggould-tri, 2018-12-19:
Notably all of the failures are on mac (thanks @jamiesnape for the catch there).

ggould-tri, 2018-12-19:
FYI @hongkai-dai as the author of #10066

ggould-tri, 2018-12-19:
Again, https://drake-cdash.csail.mit.edu/testDetails.php?test=24851276&build=1093936 , a testIKpointwise.simpleIKpointwise fail like the original description here

hongkai-dai, 2018-12-19:
Did this CI error happen right after #10066 was merged?

jwnimmer-tri, 2018-12-19:
It is a non-deterministic failure -- only some builds show a failure, some of the time.  I think all observed instances are subsequent to #10066, but I'm not sure if we've carefully gone back to find them all.  Since it's sporadic, we might have passed off some earlier failure as flakes and not documented them in this issue.

jamiesnape, 2018-12-19:
For `mac-high-sierra-clang-bazel-continuous-release`, at least, they are all subsequent.

jamiesnape, 2018-12-19:
I would say anything that merged before December 14th could be related, though I have not checked the all the nightlies yet.

jamiesnape, 2018-12-19:
I can reproduce locally on `master` with
```
bazel test --runs_per_test=50 \
    //attic/multibody:test_ik_pointwise \
    //examples/acrobot:run_swing_up_traj_optimization_test \
    //examples/pendulum:trajectory_optimization_simulation_test \
    //examples/valkyrie:valkyrie_ik_test
```

liangfok, 2018-12-20:
From @jamiesnape 's local testing, we know the symptoms predate #10066.

ggould-tri, 2018-12-20:
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-high-sierra-clang-bazel-continuous-release/95/console again

avalenzu, 2019-01-04:
Just hit this again:
- Jenkins: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-high-sierra-clang-bazel-nightly-release/48/
- [CDash](https://drake-cdash.csail.mit.edu/testDetails.php?test=25544173&build=1098776)

ggould-tri, 2019-05-24:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-high-sierra-clang-bazel-nightly-python3-release/137/ again

jamiesnape, 2019-05-24:
Is that test marked flaky?

EricCousineau-TRI, 2019-05-28:
Nope, `//bindings/pydrake/multibody:py/inverse_kinematics_test` is not flaky. The other tests are not marked either.

Should we mark all of them?

jamiesnape, 2019-05-28:
I think so. They are indeed slightly flaky and we are no closer to working out why. Assuming that they are fairly quick-running tests, I don't think there is a huge downside.

jamiesnape, 2019-05-28:
Since https://github.com/RobotLocomotion/drake/issues/10276#issuecomment-451455852 is now in part a list of HTTP 404 errors, note that Jenkins links only last 90 days. CDash links last 365 days.

jwnimmer-tri, 2020-02-03:
As of #12664, Valkyrie code no longer exists on master, which means that `//examples/valkyrie:valkyrie_ik_test` is longer a problem.

jwnimmer-tri, 2020-05-14:
I am not sure a long-lived issue is buying is much.  If we start seeing tests fail, we should deal with them immediately -- either marking them flaky, or debugging, or bisecting, or what have you.

closed 2020-05-14 by jwnimmer-tri