RobotLocomotion/drake #10288: pydrake: Reduce size occupied by modules

EricCousineau-TRI, 2018-12-20:
At fdd7553, in the `//:install` tree, `pydrake` occupies 112MB of space out of 566MB. For comparison, `libdrake.so` occupies 26MB.

This amount should be reduced, either via #8433 or by reducing the overall module count.

Will increase priority if this starts to become a bottleneck.

jamiesnape, 2019-01-15:
Does not seem unreasonable to me. Backlog?

EricCousineau-TRI, 2019-01-15:
Not yet, depending on what solution we want to take for #10408.

jwnimmer-tri, 2019-01-15:
@EricCousineau-TRI could you re-check the object code sizes, now that #10350 has merged?

jamiesnape, 2019-01-15:
> Not yet, depending on what solution we want to take for #10408.
I don't think #10408 changes anything. 112MB of 556MB is not unreasonable, and to be honest ~224MB of ~668MB is equally not unreasonable.

EricCousineau-TRI, 2019-01-15:
All sounds good; knocking down priority to backlog. Will still briefly check sizes, probably tomorrow though.

jwnimmer-tri, 2019-02-14:
If we ever pick this up again, we should try https://github.com/google/bloaty, e.g.,

  `bloaty -C full -d segments,sections,shortsymbols,symbols $(find bazel-bin/bindings/pydrake -name \*.runfiles -prune -o -name _objs -prune -o -name \*.so -print)`

jamiesnape, 2019-02-14:
Aside: Running the aforementioned McBloatface on `libdrake.so` is kind of interesting too (`drake::math::GrayCodeToInteger()` comes in at almost 20% of the size of the binary on Mac, next highest is 3%).

EricCousineau-TRI, 2019-03-23:
Posted a brief benchmark script to run from source, specific to these artifacts. Haven't used McBloatface, yet.

Briefly tried using `namespace pydrake __attribute__((visibility("hidden")))`, but that wasn't the easy fix  - size is almost exactly the same :(
```
$ git diff
diff --git a/bindings/pydrake/pydrake_pybind.h b/bindings/pydrake/pydrake_pybind.h
index cdfa1b5d1..da87675be 100644
--- a/bindings/pydrake/pydrake_pybind.h
+++ b/bindings/pydrake/pydrake_pybind.h
@@ -7,7 +7,7 @@
 // for) these binding capabilities.
 
 namespace drake {
-namespace pydrake {
+namespace pydrake __attribute__((visibility("hidden"))) {
 
 /** @defgroup python_bindings Python Bindings
 @ingroup technical_notes

$ ... # Ran benchmark above for Python 3:
121M    total
```

jwnimmer-tri, 2023-05-22:
For the record, this is a lot better now that I've been folding modules together.  I would advise we finish that first (getting to the point where we only have two shared libraries -- `libdrake.so` + `libpydrake.so`), and only after that we could investigate hidden symbols or other linker optimizations within `libpydrake.so`.

jwnimmer-tri, 2023-11-07:
The change in #20465 helped somewhat (5%).  Ongoing module-folding is on deck to continue to improve.

Also #20296 will help a lot.