RobotLocomotion/drake #10298: support symbolic random functions (Vector<Expression>)

RussTedrake, 2018-12-21:
Unlike the existing Vector<Expression>, where each element can always be evaluated separately, I think in the random case I need  to call `Evaluate(vector<Expression>, generator)` and have the same random value used for all of the outputs?

in other words... if I have
```
v = Variable(RANDOM_UNIFORM)
y1 = 5*v + 1
y2 = 3*v + 3
y = Eigen::Vector2<Expression>(y1, y2)
```

i would like to be able to call `Evaluate(y1, generator)` and `Evaluate(y2, generator)` separately, which would result in independent draws from the distribution of `v`.
but I would also like to be able to call `Evaluate(y, generator)`, and have `v` drawn only once, but used for both `y1` and `y2`.

jwnimmer-tri, 2018-12-21:
For both performance an aesthetics, I agree that we should probably hvae a function `VectorXd Evaluate(const Ref<const VectorX<Expression>>&, Generator*)` with the consolidated random variable sampling.

You could also imagine a slightly more general form: `Environment Sample(const Ref<const VectorX<Expression>>&, Generator*)` that _just_ does the sampling, and lets the user either `Evaluate` or `Substitute` or whatever they want.

soonho-tri, 2018-12-21:
@jwnimmer-tri , in #10299 , I've extended the existing `Evaluate` function which can take both of dynamic and static sized matrices.

soonho-tri, 2018-12-21:
> You could also imagine a slightly more general form: Environment Sample(const Ref<const VectorX<Expression>>&, Generator*) that just does the sampling, and lets the user either Evaluate or Substitute or whatever they want.

We already have this one: 
https://github.com/RobotLocomotion/drake/blob/1e8c81b6e9d37d89957e1bf1e51dc6273c39dfd4/common/symbolic_environment.h#L140-L144

jwnimmer-tri, 2018-12-21:
So then `Sample(exprs, &gen)` can be achieved with `PopulateRandomVariables({}, GetDistinctVariables(expressions), &gen)`.  This seems fine.

jwnimmer-tri, 2018-12-21:
Personally I'm okay if that waits until we need it.  I didn't realize those other two primitives (Populate, Distinct) already existed.

soonho-tri, 2018-12-21:
> Personally I'm okay if that waits until we need it.

Got it.