RobotLocomotion/drake #10304: Python PATH is Broken on Mac High Sierra

liangfok, 2018-12-21:
See the errors here:

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-high-sierra-unprovisioned-clang-bazel-nightly-release/35/

```
env: python2: No such file or directory
```

This is likely due to a Bazel update. An RC file likely needs the following to be added:

```
--action_env=PATH=/usr/local/bin:/usr/bin:/bin
```

jamiesnape, 2018-12-21:
Relates https://github.com/bazelbuild/bazel/issues/6648.

EricCousineau-TRI, 2018-12-21:
Would adding this to `install_prereqs_user_environment` for Mac make it work?

jamiesnape, 2018-12-21:
It would, though technically this is not a Mac issue.

jamiesnape, 2018-12-21:
I imagine the analog for Linux would be `--action_env=PATH=/usr/local/bin:/usr/bin`.

jamiesnape, 2018-12-21:
I think this also makes our own version of the `bzl` `which()` function redundant, but that is not going to break anything.

jwnimmer-tri, 2018-12-21:
> I think this also makes our own version of the bzl which() function redundant, but that is not going to break anything.

Not quite.

(1) When our `//tools/workspace` rules are used from a `bazel_drake_external`, we don't know what the action `$PATH` will be.  I think its still good to ignore whatever the user configured, at least by default.

(2) I'm not sure that `build --action_env` affects, e.g., `bazel fetch //...`.

PR #10305 (2019-01-02): bazelrc: Add platform-specific --action_env=PATH
Resolves #10304

\cc @liangfok

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10305)
closed 2019-01-02 by EricCousineau-TRI