RobotLocomotion/drake #10309: MBP/MBT methods could take model_instance = ModelInstanceIndex{} and check is_valid()

RussTedrake, 2018-12-22:
It seems like MBP and MBT have almost twice as many methods as they need to have.  Almost every method (from `num_positions()` to `GetJointByName()`) has at least two variants -- one that takes a `model_instance` and one that does not.  This starts to explode when, for instance `GetJointByName` has a templated version and a non-templated version (so 4 entry points).  I need to now add the mutable variety... so 8 entry points!!  And every one of them is repeated in MBP and MBT.  (16 copies!).  And there are lots of copies of the method documentation flying around.

Is there a reason we don't just take `model_instance` as an optional argument with a default value of `ModelInstanceIndex{}`, and then check `is_valid()` on the inside?  It would cut down the entry points by half at least?  And presumably the cost would be negligible?

EricCousineau-TRI, 2018-12-22:
Relates #10194

Agree on this front, along the lines that we should reduce the number of overloads in the public API.

(Reworded title to point to just MBP; MBT should no longer be a considered a public API.)

sherm1, 2018-12-23:
Could use `optional<ModelInstanceIndex>` for that fresh up-to-date look!

RussTedrake, 2018-12-24:
I agree that the optional version can work (have it on my branch), but I feel like it is extra unwarranted machinery, since the TypeSafeIndex already has the notion of invalid indices?  But if people agree it's more clear, I'm ok with it.

jwnimmer-tri, 2024-03-20:
For the record, `GetJointByName` is already de-dup'd.  Other things like `GetJointActuatorByName` remain overloaded.