RobotLocomotion/drake #10310: MultibodyPlant should be `final`

EricCousineau-TRI, 2018-12-22:
We have tests that have classes that inherit from `MultibodyPlant` (e.g. `AxiallySymmetricFreeBodyPlant`).

The API of MBP has enough complex behavior hidden that it seems like a bad idea to encourage inheriting from this class. We should prevent this.

sherm1, 2018-12-22:
That would be unfortunate IMO. Very useful to be able to say that something is-a MBPlant.

jwnimmer-tri, 2019-01-02:
I agree with Eric -- I don't think the type checking that comes from `MySpecialPlant : public MultibodyPlant` is worth the headaches of making the class design open.  For me, `MultibodyPlant* my_special_plant` is plenty good enough documentation (we have several instances of this formulation already in the code).

If you really _do_ need to keep it non-final, then you need to look under [System Scalar Conversion](https://drake.mit.edu/doxygen_cxx/group__system__scalar__conversion.html) at **Systems not marked as final** and repair MultibodyPlant to actually be correctly subclass-able.

sherm1, 2019-01-02:
OK, good point. I'm happy to have it be final for now until we have a pressing need. Nothing would stop us from changing it later.

EricCousineau-TRI, 2019-01-02:
Alrighty, sounds good! I'll take a pass at sealing it off sometime next week.

jwnimmer-tri, 2019-01-17:
For the record, we had a long discussion about this in #10414 -- also concluding to mark it `final`.

PR #22397 (2025-01-06): [multibody] Deprecate subclassing of MultibodyPlant
Closes #10310.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22397)
closed 2025-01-06 by sherm1