RobotLocomotion/drake #10319: Uniform random rotations (bug?)

RussTedrake, 2018-12-27:
Was curious about the mathematical justification for the uniform random rotations so I looked at the justification/reference here: 
https://github.com/RobotLocomotion/drake/blob/master/math/random_rotation.h#L20
But that reference only tells me how to sample uniformly from the sphere; nothing about rotations.  

According to 
http://en.wikipedia.org/wiki/Rotation_matrix#Uniform_random_rotation_matrices
Our implementation is actually wrong.  The angle of axis angle should not be uniformly distributed.

Proposal:
`UniformlyRandomQuaternion` should just sample from the unit sphere (4D), via four Gaussian random variables normalized.  `UniformlyRandomAxisAngle` (and others) should use the `UniformlyRandomQuaternion`.

Cc @hongkai-dai

hongkai-dai, 2018-12-27:
Yes I agree what we did (sampling a unit axis and a random angle) doesn't generate a uniform rotation.

On the other hand, I am not fully convinced that by generating a unit vector in 4D sphere, the quaternion gives us a uniformly distributed rotation. In quaternion q = [w, x, y, z], w = cos(theta/2), x = a * sin(theta/2), so`w` and `x` is not "symmetric". But by randomly sampling on a 4D sphere, I think we are treating as if `w` and `x` can be interchanged.

From that wikipedia link https://en.wikipedia.org/wiki/Rotation_matrix#Uniform_random_rotation_matrices

_We can also generate a uniform distribution in any dimension using the subgroup algorithm of Diaconis & Shashahani (1987). This recursively exploits the nested dimensions group structure of SO(n), as follows. Generate a uniform angle and construct a 2 × 2 rotation matrix. To step from n to n + 1, generate a vector v uniformly distributed on the n-sphere Sn, embed the n × n matrix in the next larger size with last column (0,…,0,1), and rotate the larger matrix so the last column becomes v_

So we can first generate a random 2 x 2 rotation matrix (by uniformly sample theta in [0, 2pi]), and then generate a 3 x 3 rotation matrix. In the end we need to convert this rotation matrix to a quaternion.

RussTedrake, 2018-12-27:
Also in the wikipedia:

> Another method uses unit quaternions. Multiplication of rotation matrices is homomorphic to multiplication of quaternions, and multiplication by a unit quaternion rotates the unit sphere. Since the homomorphism is a local isometry, we immediately conclude that to produce a uniform distribution on SO(3) we may use a uniform distribution on S3. In practice: create a four-element vector where each element is a sampling of a normal distribution. Normalize its length and you have a uniformly sampled random unit quaternion which represents a uniformly sampled random rotation.

that just seems easier.  no?

RussTedrake, 2018-12-27:
I remembered that derivation, too.  Thanks for digging it up!  I agree that Steve's pick is the best for us, too.  Not only does it have one less random variable, but it appears (to my eye) to have better numerics -- less likely to have divide by zero issues.  Ship it!

RussTedrake, 2019-03-25:
@sherm1 @mitiguy ping on this.  just saw that it's still open and no movement?  i would think it's an important and relatively quick fix?

sherm1, 2019-03-25:
I'm guessing the hardest part of this would be coming up with a unit test that fails with the current implementation but succeeds with the correct one. Any ideas on how to do that?

RussTedrake, 2019-03-26:
i think it should be relatively easy to test if it's uniform or not?  e.g. like i did for the random sources?
https://github.com/RobotLocomotion/drake/blob/master/systems/primitives/test/random_source_test.cc#L53

mitiguy, 2019-03-26:
I'll look at creating a test that:
1.  Makes the current test fail.
2.  Checks that the better algorithm succeeds.
I'll reference the books you mentioned -- stats is not my strength.

RussTedrake, 2019-03-26:
The algorithm is trivial, and we agree that it should be right.  I can PR it myself if it’s too much trouble...

sherm1, 2019-03-26:
We acquired the reference "Simulation and the Monte Carlo Method by Rubinstein and Kroese, 2017 page 108". Without Hongkai, we are short on stats chops around here -- determinism is our collective strength!

jwnimmer-tri, 2020-01-10:
@sherm1 @mitiguy ping.  If I am reading my plots correctly, Anzu's monte carlo simulation testing is experiencing non-uniform samples when trying to measure our task performance, probably due to this bug.

sherm1, 2020-01-10:
Upped priority to high. Per discussion above, the unit test is the primary holdup -- anyone who knows how to do that, please feel free to PR this.

SeanCurtis-TRI, 2020-01-10:
To use that approach, we need to bin rotations. And it doesn't seem obvious how we're going to go about doing that.

jwnimmer-tri, 2020-01-10:
My quick read of the wiki page (https://en.wikipedia.org/wiki/Rotation_matrix#Uniform_random_rotation_matrices) was that in AxisAngle representation, for each of the three angles treated individually, the CDF should be 1/π(θ − sin θ), for 0 ≤ θ ≤ π.

edrumwri, 2020-01-10:
The wikipedia page talks about the one angle, not the three components of the axis. And I don't believe you can use the CDF as a test. You'd need to be able to draw samples from particular angles to integrate the CDF.

mitiguy, 2020-01-10:
Reassigned from @mitiguy to @hongkai-dai
Hongkai and I just talked.  He is better suited for the uniform and random aspect of this.  I will work 
with him as needed on the rotation part.

PR #12562 (2020-01-18): Re-implement the uniformly random rotation.
Resolves #10319 

@soonho-tri , could you please take a look at this PR? I found that now UniformlyRandomAngleAxis<T>() doesn't work for T = symbolic::Expression. You could uncomment the line https://github.com/hongkai-dai/drake/blob/61ec160b2ad50381124e2ac7a7153d4cbbca104b/math/test/random_rotation_test.cc#L76-L77, the code throws an error
```
unknown file: Failure
C++ exception with description "The following environment does not have an entry for the variable random_uniform_0

" thrown in the test body.
```
And I checked that the environment is em [... cut]
closed 2020-01-18 by hongkai-dai