RobotLocomotion/drake #10320: Some constraint is not thread safe

hongkai-dai, 2018-12-27:
`ExpressionConstraint` is not thread safe, as it contains *mutable* `symbolic::Environment`
https://github.com/RobotLocomotion/drake/blob/f4a0b01b5f16308676529aa540e8cffcf88c0ca5/solvers/constraint.h#L902-L903

We can have two solutions

1. Add `bool is_thread_safe` to `EvaluatorBase` class. This flag is default to false, and every constraint/cost has to set it to true if the user wants to call them in multiple threads.
2. Fix `ExpressionConstraint`, we can pass a `symbolic::Environment` pointer to `ExpressionConstraint` like
```C++
class ExpressionConstraint {
 public:
  ExpressionConstraint(const symbolic::Expression& e, symbolic::Environment* const env);

  template <typename T>
  void DoEvalGeneric(const Eigen::Ref<const VectorX<T>>& x, VectorX<T>* y) const {
    UpdateEnvironment(x, env_);
    // Then evaluate the expression with this env.
    *y =  env_.Evaluate(e_);
  }

 private:
  symbolic::Expression e_;
  symbolic::Environment* const env_;
}
```
This is similar to how we pass `Context` pointer to InverseKinematic constraint, and evaluate these constraints by first updating the context. This would make `ExpressionConstraint` to be thread safe.

@RussTedrake @avalenzu @jwnimmer-tri do you have some suggestions? Which approach would you prefer?

RussTedrake, 2018-12-27:
I prefer -- both!  There are other constraints out there (e.g. the direct collocation constraint) that are also not thread safe.  I think we should systematically clean them up and set the flag to "safe" every time we bless an implementation.  But new constraints should have to opt-in to being declared thread safe.  

The only alternative I see is to fix every constraint we know about, and then document the assumption that EVERY constraint must be implemented as thread safe.

jwnimmer-tri, 2019-01-03:
I would do this:
(1) Run a benchmark, to get a baseline performance measurement.
(2) Change `environment_` to be a local variable `env` in DoEval, not a member field.
(3) Re-run the benchmark.  If it's fast enough, you're done.
(4) Profile what's slow in DoEval.  Is it rehashing, or is is malloc?
(5) If its rehashing, trying adding method `Environment::reserve`, since we know up-front how many variables we're going to need.
(6) If its reallocation, trying marking the local variable `env` as `static thread_local` (and clear it before use) so that all DoEval calls in a thread share the same temporary heap `env`.
(7) Re-run the benchmark.  If it's fast enough, you're done.

avalenzu, 2019-01-04:
TIL about `thread_local`!

jwnimmer-tri, 2019-01-09:
FYI I think `PolynomialEvaluator` has the same problem -- using mutable member fields to pre-allocate scratch storage.

jwnimmer-tri, 2020-05-14:
Relates #13099.