RobotLocomotion/drake #10327: Per-step publish events should be dispatched in Simulator::Initialize()

sherm1, 2018-12-30:
Writing documentation in #10321 for per-step events I realized that the very first trajectory value never gets published, potentially causing the same problem fixed in #9766 for timed events.

The per-step publish events should get dispatched along with the periodic publish events at the end of Initialize(). The per-step discrete and unrestricted update events are already dispatched correctly at the start of the first StepTo().

edrumwri, 2018-12-30:
The behavior you're describing is not obviously incorrect: I'm not sure which "trajectory value" you're referring to, and executing a per-step publish at initialization would seem to violate the Principle of Least Astonishment™. Please provide more information about the particulars of the problem that you experienced so that I can either agree with you that it's a problem or argue otherwise.

sherm1, 2018-12-30:
Here's what I said in trying to document per-step events for the sugar methods in PR #10321:
> Per-step events are useful for taking discrete action at every point of a
simulated trajectory (generally spaced irregularly in time) without
missing anything. For example, if you want to record in a state variable
the minimum or maximum value achieved so far by some quantity, a per-step
event guarantees you will see it, while a periodic event could miss the
extremum.

As soon as I wrote that I realized it wasn't true since a per-step publish event would miss the t=0 value. Here's a way to reason about this in PLA™ terms: suppose a user is simulating with fixed-size, 1ms steps, and has a periodic publish event with period 1ms, offset 0. We publish the trajectory at 0, 1, 2ms, etc., that is, every step. We know that's the right behavior (after much agony)! Now said user notices that we have a per-step event, which seems better since there is no period to be kept in sync with step size (which might have to change). So s/he replaces the periodic publish event with a per-step publish event. Now the published output is at 1, 2, 3ms -- the 0 publish is gone. Are they surprised? Also, how about the fact that per-step discrete and unrestricted update events _do_ occur at time 0 -- only publish events neglect 0. That also seems surprising to me. What do you think?

sherm1, 2018-12-31:
Glad you like how I described per-step events -- too bad it's currently a lie though!
I don't yet see how the situation is any different than for the periodic events you fixed in #9766. We are missing one of the publishes. Seems like a bug and something we should fix asap while we still remember how all the event sequencing works. Then my description would be accurate.

I've always used "step" in a variable-step context and I believe that is common practice. Matlab uses it, for example. They actually go further with a concept of ["minor step" vs "major step"](https://blogs.mathworks.com/pick/2011/06/10/minor-time-step-logging).  Our "per-step" corresponds to a "major step" there; minor steps are the throwaway intermediate evaluations in multistage integrators. The notion that a simulated continuous trajectory is represented by discrete steps of varying density is relevant to users, especially since the density is related to the local complexity of the trajectory (for example, a plot generated with periodic sampling will miss local high-frequency behavior). What concept would you propose rather than "step"?

(Happy to discuss f2f when we get back.)