RobotLocomotion/drake #10329: Doubling of SceneGraph public API

SeanCurtis-TRI, 2019-01-02:
@EricCousineau-TRI  (inspired by #10309)

`SceneGraph` is risking a doubling of *its* public API as well. The [two vairants](https://github.com/RobotLocomotion/drake/blob/972825e4cd3c224b6e03d379feeaa476b777ee2f/geometry/scene_graph.h#L184-L193) for modifying the model vs a context double the API.

The *primary* cause for this is the practice of passing a context *first* when it is an argument which kills the possibility of using an optional parameter. And making it *literally* an `optional` would lead to stupid `{}` parameters in most normal invocations.

So, before this API-doubling property gets arbitrarily extended, I'd like to solicit opinions.

sherm1, 2019-01-02:
Don't these methods also differ in whether they are const or non-const? That seems like a significant enough difference to warrant a pair of methods.

SeanCurtis-TRI, 2019-01-02:
D'oh!  Too much copy and paste. They *should* be const... Thanks for calling that out.

EricCousineau-TRI, 2019-01-04:
Hm... Is there any way in which you can modularize these operations, even it costs an extra step?

e.g. if you currently have:
```
scene_graph->Something();  // No context
scene_graph->Something(&context);  // Aha!
```
could you instead do:
```
scene_graph->state().Something();  // Prefer `state` over `get_mutable_state()` and just `const`ness to infer?
scene_graph->state(&context).Something();  // Aha!
```

Is the underlying object just a `QueryObject` or `GeometryState`?
Is there any difference in the actual operations that happen, or is just some indirection for sugar?

SeanCurtis-TRI, 2020-09-25:
I'm inclined to close this.

1. I'm against leaking `GeometryState` into the public API.
2. I can't make `Context` optional because we have a convention that when a `Context` is provided its the first parameter.

So, other than noting a doubling, I don't anticipate this getting resolved any time in the foreseeable future.

closed 2020-09-25 by SeanCurtis-TRI

EricCousineau-TRI, 2020-09-25:
1. I mean, it can be whatever interface that publicly pipes `GeometryState`, if you're concerned about things that `GeometryState` may want to expose internally.
2. That could be worked around with pointers? I don't see that as a valid objection.

That being said, in terms of "what's important and when will it get done?", I'm OK with this being closed.
I just forsee it will beget additional work, esp. in terms of #13445 and how much new API needs to get added b/c none of this wrapped in a single interface.

SeanCurtis-TRI, 2020-09-25:
I'm not entirely sure I understand your statements.

1. "whatever interface that publicly pipes `GeometryState`"? So, you're suggesting a *third* class? Something that sits between SceneGraph and GeometryState? That's returned by the putative `state()` method? That could work. All of `SceneGraph`'s forwarding APIs get passed to the new class, and that class only has to have one variant on each call, because it relies on `SceneGraph` to have returned the right version of it (model vs context). The danger is if someone hangs onto an instance of the new guy. Right now, `SceneGraph` doesn't *return* anything that could be persisted and mucked about with. And that's nice.
2. I'm not sure what you mean by "worked around with pointers"? Do you  mean simply have it *not* be an optional parameter (in the C++ syntax) but allow it to be `nullptr` and remain as the first argument such that people explicitly invoke it either with `nullptr` or `{}`?