RobotLocomotion/drake #10333: cdash: Job results sometimes do not appear?

EricCousineau-TRI, 2019-01-03:
In the following job:
https://drake-jenkins.csail.mit.edu/job/linux-bionic-gcc-bazel-experimental-release/851/

The CDash output does not show up:
https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-linux-bionic-gcc-bazel-experimental-release-851&field2=buildstarttime&compare2=84&value2=now

I've seen this happen occasionally, but had not recorded it. Any chance we know why this happens?

EricCousineau-TRI, 2019-01-03:
Hup, seems that it's just a timing thing? Will close this for now.

closed 2019-01-03 by EricCousineau-TRI

jamiesnape, 2019-01-03:
Relates https://github.com/jenkinsci/ansicolor-plugin/issues/145 ?