RobotLocomotion/drake #10334: jenkins: Colored output obscures console and plain text views

EricCousineau-TRI, 2019-01-03:
Related #10333; however, in this case, the colored output in Jenkins `Console Output` still has some bits obscured:

> [image]

Additionally, the plain text is also difficult to read:

> [image]

It'd be nice to have the `Console Output` show up correctly.
Additionally, if there is an easy way to scrub color escape sequences from text, it'd be nice to have some sort of purely plain text view available.

jamiesnape, 2019-01-03:
Relates jenkinsci/ansicolor-plugin#145 ?

jamiesnape, 2019-01-03:
I will try to downgrade the plugin, but it requires a restart of Jenkins so it won't happen particularly soon.

EricCousineau-TRI, 2019-01-10:
Resolved by downgrading the plugin.

closed 2019-01-10 by EricCousineau-TRI