RobotLocomotion/drake #10339: to enable public DeclareSystemConstraint -- support SystemConstraint  scalar conversion

RussTedrake, 2019-01-03:
Related to #10290 and #10291.

Currently `SystemConstraint<T>` takes a function pointer in its constructor.  Like output ports, etc, this means that the `DeclareSystemConstraint` needs to be a protected method, because we can only transmogrify `SystemConstraints` by having the `System` know how to create them again.

This workflow breaks down, for instance, when we want to declare constraints for a `MultiBodyPlant` loaded from SDF.  (We don't want users to have to derived from `MultiBodyPlant`).

Proposal (from me and @sammy-tri after a good discussion):

- `SystemConstraint::SystemConstraint(functor)` => `SystemConstraint::SystemConstraint(const System<T>& system)`.
- To implement a new constraint, authors will derived from SystemConstraint and overload the `EvalConstraint` method.  (just like a generic constraint in `MathematicalProgram`)
- teach `SystemConstraint` to do scalar conversion, by offering `SystemConstraint<U>::ToScalar(const SystemConstraint<T>& constraint, const System<U>& system)`

- Probably we'll want to offer some canonical constraints (e.g. linear function of the numeric state, linear function of the numeric parameters, ...) e.g. to even maintain the current the `VectorBase` constraint workflow.

cc @hongkai-dai

jwnimmer-tri, 2019-01-03:
> Currently SystemConstraint<T> takes a function pointer in its constructor. Like output ports, etc, **this means that** the DeclareSystemConstraint needs to be a protected method, ...

This does not follow.  The calc function does _not_ need to be a bound member function of a System.  The user can supply any functor they like.

It sounds like the actual problem here is that you want MultibodyPlant to offer an API to is users, where they can _add_ their own constraints to the plant, and then those constraints should remain intact as the plant transmogrifies.

Since the set of allowed scalar types for MultibodyPlant is closed, we can implement this easily by having the user call an `MultibodyPlant::AddConstraint` method, passing in a generic lambda.  The implementation will instantiate the lambda for the required scalars, and store the result as a member field of the plant, so that it can call `System::AddConstraint` with the right functor.

I would do this first just for MultibodyPlant.  If we find we need it more widely, we can factor it into a reusable has-a helper (like we did for ModelValues, so something like a ModelConstraints shim).

I think the proposal in the top post is very difficult and overly complicated, once you start figuring out how to implement ToScalar (you end up with type tags, and lots of machinery).

jwnimmer-tri, 2019-01-04:
For example, a bounding box constraint on a model's positions would look something like this:
```c++
auto calc_constrained_value = [my_model_instance](
    const auto& plant_system, const auto& plant_context, const auto* result) {
  *result = plant_system.GetPositions(plant_context, my_model_instance);
};
VectorXd lower, upper = ...;
mbp.AddUserConstraint("name", lower, upper, constrained_value);
```

RussTedrake, 2019-01-04:
i did not realize you could create a lambda who's types were lazily bound (by the caller).  that's awesome.  Perhaps we could have / should have used that for the distribution workflow, too.  (I'm sure that we still can, and could easily offer both).  Would love to see this play out for MBP.

RussTedrake, 2019-01-10:
The simplest example would be a fixed point constraint (which should work even at the System level).

CalcTimeDerivatives should return zero.
Any velocities in the context must be zero.
All discrete updates should be xnext = x

This would compute what i need for the mbp clutter (but i will also make a more specialized/efficient version for mbp)

jwnimmer-tri, 2019-01-11:
In #10394, I have `GTEST_TEST(ExternalSystemConstraintTest, FixedPoint)` now.

jwnimmer-tri, 2019-01-22:
@RussTedrake I hope that #10394 has resolved all of the required use cases.  Can you check and see if the framework is sufficiently capable and usable now?  We might need more examples, or more sugar, or even more features.

RussTedrake, 2019-01-22:
I'm sure that we will build up more features, but based on my reading of that PR I think it's safe to close this issue (feel free to reopen if you disagree).

closed 2019-01-22 by RussTedrake