RobotLocomotion/drake #10348: VectorSystem should get new event sugar methods

sherm1, 2019-01-07:
VectorSystem provides a simplified interface to LeafSystem. In #10321 LeafSystem gained some sugar for declaring periodic, per-step, and initialization events. VectorSystem should get parallel ones, except with handler callbacks that take Eigen vector quantities rather than Drake objects.

Q: Are there other APIs we've introduced into LeafSystem that ought to have VectorSystem analogs?

sherm1, 2019-01-07:
@edrumwri pointed out that VectorSystem doesn't provide input/output port naming (though the defaults should be fine there).

Per discussion in #10346, it's conceivable that we have improved LeafSystem and other framework classes so much that VectorSystem doesn't provide as much simplification as it used to. It does require active maintenance to keep it up to date with changes to LeafSystem so we should make sure it is still a win to have it around.

Are there any VectorSystem fans out there who could weigh in?

jwnimmer-tri, 2019-01-07:
I think its unlikely that the target users for VectorSystem are subscribed to Drake's issue firehose.

jwnimmer-tri, 2024-03-16:
https://github.com/RobotLocomotion/drake/issues/20989#issuecomment-2002129534

closed 2024-11-01 by jwnimmer-tri