RobotLocomotion/drake #10352: Some Unicode characters on Mac do not render properly

rcory, 2019-01-07:
Moving over an email thread from a few months back to track this issue. 
/cc @RussTedrake @sherm1 

> There are an appreciable number of unicode subscripts that don't render on the two Macbooks we tested on. These include lower case h, k, l, m, n, p, s, t. We also found that Firefox seems to do a better job at rendering known unicode characters than Safari or Chrome (specifically, there are superscripts that are aligned poorly in Safari/Chrome). We also convinced ourselves that this is a system wide font issue (not just browsers), that affect all text editors we tried (e.g., vim, emacs, TextEdit, and the Mac mail client). My suspicion is that there are probably other rendering issues lurking throughout the documentation when viewed on a Mac.

> The recommendation for Mac users would be:
> 1. Install the Symbola font (font linked here, taken from this website). This only takes a few seconds.
> 2. Verify the Symbola font is enabled. Do this by running the "Font Book" application (go to spotlight, search for "Font Book" and open it). On the left column, select "User", right click on the "Symbola" font and if "Enable Symbola Family" appears as an option, select it. Otherwise, it was automatically enabled by the installation.
> 3. Restart your web browser
> 4. For best results, use Firefox to browse Drake doxygen

> Here's another surprising data point. On a mac you can invoke the system character map by doing "cntrl+cmd+spacebar". By default, the unicode tables are not listed, but you can add them by clicking on the small gear on the top left of the application, select "Customize List...", scroll down to "Code Tables" and expand the menu, and select "Unicode". On the left panel select the "Unicode" character set, which allows you to see the currently recognized unicode characters.

> If you scroll down to the Unicode set starting at 2070 (Superscripts and Subscripts) you'll see that out of the box Mac doesn't recognized the subscripts I pointed out in my earlier email. After installation of the Symbola font, the subscripts show up. It seems that those unicode characters are just not there to begin with.

> Before installing the Symbola font:
[image]

> After installing the Symbola font:
[image]

sherm1, 2019-01-07:
Been broken since [at least 2014](https://discussions.apple.com/thread/6030389).

soonho-tri, 2019-01-07:
FYI, installing http://www.quivira-font.com/files/Quivira.ttf can help as well (as mentioned in the "at least 2014" link above and https://github.com/RobotLocomotion/drake/pull/5197#issuecomment-281530638).

jwnimmer-tri, 2019-01-07:
\CC @jamiesnape in case you have thoughts.

jamiesnape, 2019-01-07:
Yes, for whatever reason there are very few fonts for Mac that cover the range of unicode used in Drake. Not sure it is an actionable bug unless someone wants to add a note to docs or we move away from troublesome characters.

jwnimmer-tri, 2019-01-07:
It came up again recently in the PR discussion for #10346 for `examples/simple_continuous_time_system.cc` and related, where the subscripts are being ever-more-widely used.

jamiesnape, 2019-01-07:
Personally, I have no trouble reading unprocessed LaTeX in the source code, and obviously it is more readable than a missing character symbol, so there is probably a choice to be made then.

This is not a new problem and I have tended to just shrug and move on as Drake is so heavily invested in Unicode and Mac is a minor platform. Ideally it would not be that way, of course.

sherm1, 2019-01-09:
From discussion in #10346:
> Russ: I think the most satisfying solution so far would be x_n[index] if mac supported those fonts. Might we insert something like font-face or the popup solution into our documentation pages?
https://stackoverflow.com/questions/19567331/missing-fonts-on-web-pages

> Evan: +1 on @RussTedrake's last proposal: let's figure out how to address this population of users with broken fonts rather than restrict ourselves to a unicode subset (which itself requires workarounds for some subscripts and superscripts).

Current plan is to work around this problem in #10346 using LaTeX-like `x_n` and `x_{n+1}` syntax for now and sort this out separately.

sherm1, 2019-02-05:
[`@font-face`](https://developer.mozilla.org/en-US/docs/Web/CSS/@font-face) looks promising as a way to automatically install missing fonts on the Mac.

Does anyone know how we could use that to get our Doxygen to render properly on a virgin Mac?

jamiesnape, 2019-02-05:
We can specify the HTML header and CSS file, so should be easy enough for the purposes of the online documentation.

sherm1, 2019-02-05:
Would you have time to make a PR for that, @jamiesnape ? I have an unmodified Mac I could try it on (@RussTedrake has one also).

jamiesnape, 2019-02-05:
Yes.

jamiesnape, 2019-02-07:
DejaVu Sans looks to be the best bet (closest to the current fonts used by Doxygen) and Ubuntu users may already have it locally so they don't have to wait the extra few milliseconds for a font to download.

sherm1, 2019-02-07:
BTW @rcory recommended installing "Symbola" (above in this thread). Not sure if one is better than the other.

jamiesnape, 2019-02-07:
Symbola is serif and looks pretty old fashioned. Quivira is also serif.

PR #10628 (2019-02-14): Use DejaVu Sans fonts for Doxygen and pydrake Sphinx documentation
Closes #10352. FYI @sherm1.

Note Ubuntu users can optionally use local versions of the fonts via `sudo apt install fonts-dejavu` and Mac users can optionally use local versions via ` brew tap homebrew/cask-fonts && brew cask install font-dejavu-sans`.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10628)
closed 2019-02-14 by jamiesnape