RobotLocomotion/drake #10373: geometry: Try to reduce scope for IDs

EricCousineau-TRI, 2019-01-09:
Follow-up from this comment: https://github.com/robotlocomotion/drake/pull/10351#pullrequestreview-190478389

At present, may things in `geometry` are singletons. For the counters that are `int32`, I'm not super super worried about; however, I am most worried about rendering labels (which are `int16`), as it seems like we may easily reach the point of exhausting these if we're doing a simulation-factory-type Monte Carlo.

In terms of priority:
1. Label IDs
2. Other IDs

Proposed steps:
- Hash out design doc of what scoping would look like
  - If it makes sense to all types of labels in one fell swoop, then it sounds good;
  - Otherwise, focus on a solution specific to label IDs if/when we run into this issue.
- Profit.

This will not be an immediate issue until the color palette has its artificial limit removed.

jwnimmer-tri, 2019-01-09:
I wrote in the PR.  I don't think we should wait to fix this.  We are adding an API that promises to allocate 16-bit IDs globally (which we cap at about 11 bits anyway, if I'm reading the PR correctly) and turn them into 16-bit ground truth images on an output port.  As you note, we will run out of labels immediately in any repeated simulation that uses labels for anything ever.  I don't think we should merge the broken API only to retract it later.  Let's choose the correct label design, and then merge that.  I don't think the other GeomtryId is related at all.  We don't render those IDs into pixels.  We do render labels into pixels.

SeanCurtis-TRI, 2019-05-16:
The core of this issue is `RenderLabel`s. I believe this was all rehashed and implemented in PR #11042. I believe we can close this. Votes?

EricCousineau-TRI, 2019-05-16:
Aye! Lemme change the title to better reflect that.

closed 2019-05-16 by EricCousineau-TRI