RobotLocomotion/drake #10375: Expose inboard_frame, outboard_frame in RevoluteJoint

hongkai-dai, 2019-01-09:
Currently `RevoluteJoint` wraps ` RevoluteMobilizer`, but doesn't expose `RevoluteMobilizer`. There are some information in `RevoluteMobilizer` that can be useful. For example, the `inboard_frame()`, `outboard_frame()` function.

EricCousineau-TRI, 2019-01-09:
I think you also wanted some proxy to some form of `CalcAcrossMobilizerTransform` as well?

FTR, this presently doesn't block work, as it can be hacked around by getting the internal tree and accessing the mobilizers there.

\cc @amcastro-tri @sherm1

hongkai-dai, 2019-01-09:
> I think you also wanted some proxy to some form of CalcAcrossMobilizerTransform as well?

For my own project, I don't need this function `CalcAcrossMobilizerTransform`, but it would be nice to expose the function as well.

amcastro-tri, 2019-01-16:
Joints only provide `Joint::frame_on_parent()` and `Joint::frame_on_child()`. However I believe @hongkai-dai what you want is not the physical graph information (Joint) but more like the model tree information (Mobilizer), since you used the inboard/outboard terminology. 
We are working with @sherm1 on providing this information in #10068. Once we finish that work you be able to ask any sort of tree information to a given MBP.

hongkai-dai, 2020-09-15:
`RevoluteJoint` has `frame_on_child` and `frame_on_parent`.

closed 2020-09-15 by hongkai-dai