RobotLocomotion/drake #10383: Support multithreaded evaluation in the System Framework

sherm1, 2019-01-10:
Per discussion with @RussTedrake at the Ann Arbor offsite and today, we would like to support parallel execution in a system Diagram, likely via asynchronous evaluation of input ports with direct support from the caching system. Russ has some notes on this topic which he'll add here, and I have these from the offsite:

Sherm's notes from offsite:
- The idea is that the robot is executing a plan while the planner is working on a replan, in different threads. When the new plan is ready, the robot should switch to that plan. But it shouldn’t have to wait for the indeterminate amount of time it will take for planner to come up with something.
- Possible implementation: EvalInputPortIfReady() – i.e., robot polls planner. In the planner subsystem, computation would have been initiated [how? By request to evaluate its output? By separate output port?] and runs in its own thread or threads. When a plan is available output port is marked “ready” [should that be supported by the cache? Maybe better as an atomic int in the output port itself?]. When EvalInputPortIfReady() is called on an async output port that is “not ready” it returns immediately. Otherwise it performs the ordinary output_port.Eval().

RussTedrake, 2019-01-10:
I have a slightly different version that I like better right now.  

My motivation / justification:  we would like to be able to execute subcomponents of a complicated/expensive diagram in parallel.  Given our "pull" architecture, where evaluating events/ports is what triggers all computation, i think the only natural place to do this is inside a system's Calc___ method.

In particular, I prefer adding an additional method `EvalInputPortNonblocking` or `EvalInputPortAsynchronous` or `BeginEvalInputPort`....  this would pull on the associated output port to trigger it's evaluation, but return immediately.  The Calc method would continue, and the author would call the original `value = EvalInputPort` method to obtain the result.  So my code might look like:
```
MyCalc(...) {
  for i=0:5, BeginEvalInputPort(i);
  for i=0:5, input[i] = EvalInputPort(i);
}
```
To allow the framework to compute the 5 input ports in parallel.

This all works (easily?) if caching is turned on always for every output port.  The caching mechanism could also act as a mutex.  @jwnimmer-tri thought that might make it too easy for the user to write themselves into a deadlock -- @jwnimmer-tri can you comment?

One other point: we could also offer a generic `MultithreadingPassthroughSystem` that takes some number of inputs and offers those same inputs as outputs.  Whenever any of the output ports is evaluated, it calls `BeginEvalInputPort` on *all* of the input ports, and then returns just the one output.  This might be a way to add multi-threading into an existing diagram easily, and visibily at the level of the Diagram.

jwnimmer-tri, 2019-01-10:
An improved formulation of Russ's example would use `std::future`:
```c++
void MySystem::MyCalc(context, ...) {
  // Asynchronously begin computing the inputs.
  auto input0_future = std::async(std::launch::async, [this, &context]{
    return this->EvalEigenVectorInputcontext, 0);
  });
  auto input1_future = std::async(std::launch::async, [this, &context]{
    return this->EvalInputValue<MyBigData>(context, 1);
  });
  // ... more inputs ...

  // Gather the inputs.
  Eigen::VectorBlock<const VectorX<T>> input0 = input0_future.get();
  const MyBigData* input1 = input1_future.get();
  // ... more inputs ...

  // Compute using the inputs.
  // ...
}
```

If we find the async evaluation becoming commonplace, we can add `EvalInputValueAsync` (and etc.) to mirror `EvalInputValue` (and etc.) with the `std::future` return type to sugar up the `std::async` use.  However for an early experimentation period, I think it's okay to call `std::async` manually in the Calc methods.

So far though, that doesn't address data races.  If two async inputs depend on the same out-of-date cache entry, then they could race on writing to it as they freshen it.  That's what Russ was getting at with his comments about locking.

For a given System, we know that the dependencies of any computation form a DAG (i.e., no cycles).  So, we could lock each out-of-date computation's storage before beginning its compute (so that we know the writes during the compute are race-free), and we should have a total ordering of all locks used by the thread (which is the way we prove lack of deadlocks).  However, the calc_function implementations supplied _by the user_ still might take locks.  I think the proof of why we still will never have deadlocks becomes much more difficult in that case.  We would need to have some rule constraining what the user calc_ function are allowed to do -- something like you can't hold a lock while calling into other code.

I also worry about the cost of putting a mutex-take around every write in a computation.  It would be nice if we could somehow do that in less-than-every instance of updating a CacheEntryValue.  Possibly by making the lock guard a whole LeafContext excluding its outputs, instead of every cache value?

There are also other whole families of designs we could use.  We could have a System that was _notified_ anytime its dependencies change value, and then optimistically / asynchronously began recomputing new values.  We could have a scheduler that used the compute DAG to "push" computations in different threads (knowing, though its understanding of the DAG, that at most one thread has write authority to any single cache value).  We could use the Event system to kick off computation that only appears on an output port later (akin to a delay line) -- in fact that is how many expensive-to-simulate Sytems (like image / depth capture) already work: by the time your controller receives the sensor data, it's already (tens of?) milliseconds later.

In any case, I still have to ponder this all a bit more, but I wanted to get those early thoughts down.

I also think that _really_ need to be more specific about the goal here.  We need a concrete use case where computations are way too slow now, and where async it likely to make it much faster.  (Ideally, we have a few distinct kinds of uses in mind as we work though designs.)  Some of the trade-offs will only become clear by looking at specific situations.

jwnimmer-tri, 2023-05-19:
See also #19437 where I plan to prototype a LeafSystem that internally schedules expensive calculations as async tasks within its internal state, by using timed events.