RobotLocomotion/drake #10385: python 2+3: Figure out where cache misses happen

EricCousineau-TRI, 2019-01-10:
Slack convo: https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1546466073000800

TODO: Add details

jwnimmer-tri, 2019-01-10:
Why is this a relevant enough priority to even merit an issue?  My first reaction is that is even-less-than-baclokg.  I think the number of developers who switch between py2 to py3 is very small (and will soon go to zero), and when using `--disk_cache` is likely not that bad in the first place.  Is the goal here to have the CI cache footprint smaller or something?

EricCousineau-TRI, 2019-03-07:
BTW @jamiesnape has been hitting some pain points on this when provisioning. Significant (but not huge) pain point due to the latency.