RobotLocomotion/drake #10392: Drake Visualizer does not draw LCMGL gemoetry

mpetersen94, 2019-01-10:
## Description

When LCMGL geometry is published, a folder will pop up under the Scene Browser of Drake Visualizer labeled "LCM GL" but no geometry will be drawn in the visualizer.

## Reproduction

Run `__main__` of the [LCMGL library](https://github.com/RobotLocomotion/libbot2/blob/drake/bot2-lcmgl/python/src/bot_lcmgl/lcmgl.py) with a Drake Visualizer instance running.

## Debug Information

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

Ubuntu 16.04

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

The distributed tools - `drake-visualizer` to be specific

* Are you (a) building `drake` from source, or (b) from a binary release?

From source (but I'm not sure that matters with this issue)

* What version of `python` are you using?

+ which python2
/usr/bin/python2
+ python2 --version
Python 2.7.12

* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?

+ which bazel
/usr/bin/bazel
+ bazel version
Build label: 0.19.2
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Mon Nov 19 16:25:09 2018 (1542644709)
Build timestamp: 1542644709
Build timestamp as int: 1542644709
+ which cmake
/usr/bin/cmake
+ cmake --version
cmake version 3.5.1

    * If you are using Bazel, please provide the host settings in your present
    project workspace:

+ capture_cc_env=external/drake/tools/cc_toolchain/capture_cc.env
+ source external/drake/tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/usr/bin/gcc
+ /usr/bin/gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609

* If you are building `drake`
[from source](https://drake.mit.edu/from_source.html):

    * What git commit of `drake` are you using?

        cef0661

    * Are you building `drake` with Bazel or CMake?

        Bazel

    * Are you using `drake` directly, or consuming it as an external?

        Problem found when consuming as external but also exists directly.

jamiesnape, 2019-01-11:
When I run the `__main__` with `drake-visualizer` from `RobotLocomotion/director` `master`, I get:
```
ERROR, unsupported client request BOT_LCMGL_TEX_2D
lcmgl unknown opcode 0
...
```
and then many more lines `lcmgl unknown opcode N` for some value of `N`.

jamiesnape, 2019-01-11:
Running the example posted on Slack:
```
import bot_lcmgl
import lcm
import time
l = lcm.LCM()
g = bot_lcmgl.lcmgl("LCMGL Test", l)
for ii in range(100):
    g.glColor3f(0, 1, 1)
    g.sphere(0.01 * ii, 0, 0, 0.05, 20, 20)
    g.switch_buffer()
    time.sleep(0.01)
```
I don't get an error, but I also don't get any visualization.

jwnimmer-tri, 2021-11-10:
Closing out as "backlog unlikely to reach an actionable state within a reasonable timeframe".  We're not going to invest in drake-visualizer support for lcmgl anytime soon.

closed 2021-11-10 by jwnimmer-tri