RobotLocomotion/drake #10402: geometry_visualization: Insufficient information for distinguishing multiple model instances

EricCousineau-TRI, 2019-01-14:
Relates #10293

@calderpg-tri is unable to visualize geometries using RViz due to lack of introspection and the fact that `BuildLoadMessage` does not incorporate model-instance-level information.

Specifically, he uses the `meshcat`-type approach, of getting the message from `DispatchLoadMessage`, then crawling through the objects to reconstruct stuff. Because `GeometryState::frames_` is unordered, he can't work around this by unique-ifying names based on the ordering coming out of `DispatchLoadMessage`.

I'd suggest that first, we try to resolve #10293, which should resolve this issue as well.

IMO, I'd like the visualization methods not prepend the source name, but instead use model instances themselves. If we need more unique instances, then we should address that in another way, rather than relying on adding GUID source names.

jwnimmer-tri, 2019-01-14:
Possibly choosing to parse `lcmt_viewer_load_robot` in order to bind in additional visualizers is where the whole story went wrong.  Instead, could you use `SceneGraphInspector` directly (possibly with a bit more sugar for matching its IDs into MBP model instance IDs)?

EricCousineau-TRI, 2019-01-14:
> Possibly choosing to parse `lcmt_viewer_load_robot` in order to bind in additional visualizers is where the whole story went wrong.

Yup, definitely.

Was not aware of the level of info we had in `SceneGraphInspector`; will check.

EricCousineau-TRI, 2019-01-14:
Per f2f with @jwnimmer-tri, will close this as a Drake issue as it's focused too much hacks 'n such. Will create an issue for the underlying issue (lack of Shape introspection).

closed 2019-01-14 by EricCousineau-TRI