RobotLocomotion/drake #10403: geometry: Provide shape type introspection

EricCousineau-TRI, 2019-01-14:
At present, there's no way to introspect what shapes are in a `SceneGraph`; at present, usages like `meshcat_visualizer` (and some usages of RViz translators, such as Anzu) call `DisplatchLoadMessage`, and then inspects the corresponding LCM message as a workaround.

\cc @calderpg-tri

SeanCurtis-TRI, 2019-01-14:
FTR I wouldn't use connecting to alternative visualizers as a motivation. There is a plan/intention to provide visualizer-ports to `SceneGraph` (an initialization port and an update port) as per #8473. Also, this issue is a repeat of #9128.

EricCousineau-TRI, 2019-01-14:
Aye, it is a duplicate. Closing.

(I guess I had searched for "geometry introspection", which that issue didn't show up...)

closed 2019-01-14 by EricCousineau-TRI