RobotLocomotion/drake #10408: sphinx docs say "Python 2.7 only"

RussTedrake, 2019-01-15:
https://drake.mit.edu/python_bindings.html#python-bindings-binary

should be updated to acknowledge official support for Python 3.

EricCousineau-TRI, 2019-01-15:
Actually, the current docs are correctly worded:
> Python 2.7 is currently the only supported version for the bindings supplied **by the binary packages**.

Changed title to better reflect this. (Have discussed this with Jamie before.)

jwnimmer-tri, 2019-01-15:
In the meantime, we could amend that document to say something like "Python 2.7 is currently the only supported version for the bindings supplied by the binary packages.  To use Python 3.x, you must compile Drake from source."?

RussTedrake, 2019-01-15:
can you help me understand the reason we don’t have python 3 in binaries?

EricCousineau-TRI, 2019-01-15:
Just because we didn't capture it in #8352; was in-source support, did not capture / prioritize binary distribution yet (immediate use cases were building from source). Can raise priority on this issue if so desired.

jwnimmer-tri, 2019-01-15:
Having a single binary package seems easier for everyone?  Any reason not to (other than file size, which at least Jamie and I agree is not enough of a difference to matter)?

However, I am not sure how easy it would be to implement?  It looks like some files under `drake/lib/cmake/**` (i.e., cps-related details) vary between the python2 and python3 drake installs.  It is possible to alter those files so that a single copy can meet all needs, or do we have to install two copies to different paths, or ...?

The `*.cmake` changes are the only gotchas I see on first glance, but we should probably diff the two installs to track down any additional discrepancies, if we decide to try the single-package route.

jamiesnape, 2019-01-15:
The CMake differences are certainly not essential (typically CMake projects would not have those details). Personally I would remove them. It is not difficult for the user to derive the `PYTHONPATH` and Python version that they are using.

jamiesnape, 2019-01-15:
Basically, we can remove:
https://github.com/RobotLocomotion/drake/blob/b9e9c4fa094bfa11872e44ae6e5cc34df510709b/tools/install/libdrake/drake.cps.in#L46-L59
and
https://github.com/RobotLocomotion/drake/blob/b9e9c4fa094bfa11872e44ae6e5cc34df510709b/tools/install/libdrake/drake.cps.in#L120
and everything still works.
The user just has to derive how to set their `PYTHONPATH`, which is one line of documentation.

EricCousineau-TRI, 2019-01-15:
Sounds good to me. Per discussion in #10288 and here, I'm good with us packaging both in the same binary. We could indicate what Python versions the archive contains in the installation docs.

jwnimmer-tri, 2019-01-15:
(Trimming back the `drake.cps` is fine by me.  I think the `pybind11.cmake` also had lots of python details; can we trim that too?)

In any case, I think we can leave it to @jamiesnape to chase down these details in preparing the feature.

RussTedrake, 2019-01-15:
Thanks all.  I’ll be trying python3 in my class starting feb.  the primary distribution is via docker (so could use source), but binaries would be nice.

jamiesnape, 2019-01-15:
>  I think the pybind11.cmake also had lots of python details; can we trim that too?

Should already have happened in https://github.com/RobotLocomotion/drake/commit/dd466e07c82f0d0919ea99868bac909943f0f855.

EricCousineau-TRI, 2019-01-31:
Per meeting with Jamie, should be ready by Tuesday.

jamiesnape, 2019-02-05:
Current nightly builds have Python 3 bindings. Just needs a minor documentation update in Drake.

PR #10585 (2019-02-05): Remove caveat about lack of Python 3 in binary packages
Closes #10408.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10585)