RobotLocomotion/drake #10411: py symbolic: DeprecationWarning not raised by `e_xv == e_yv`?

EricCousineau-TRI, 2019-01-15:
Example failure: https://drake-cdash.csail.mit.edu/testDetails.php?test=25813237&build=1103482
Snippet:
```
FAIL: test_relational_operators_nonzero (symbolic_test.TestSymbolicExpression)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./bindings/pydrake/test/symbolic_test.py", line 528, in test_relational_operators_nonzero
    value = (e_xv == e_yv)
AssertionError: DeprecationWarning not raised
```

My guess is that this machine has a later version of NumPy in which the warning message has changed from what we're filtering for:
https://github.com/RobotLocomotion/drake/blob/14f5b3960/bindings/pydrake/common/deprecation.py#L175-L179

SeanCurtis-TRI, 2019-01-15:
The test in question is using Numpy 1.16.0. A sampling of half-a-dozen other mac tests indicate that they all ran against 1.15.4_2.

EricCousineau-TRI, 2019-01-15:
Seeing if I can repro with `virtualenv` on Bionic.

PR #10412 (2019-01-15): py symbolic: Filter updated warning message for == comparison
Resolves #10411

Tested locally using base commit from #10411:
> Can reproduce on Bionic with [this commit](https://github.com/EricCousineau-TRI/drake/commit/676d68efe997c897eb3521470eccbd5caee03706), and this command:
> ```
> $ source tmp/venv_setup.sh
> $ bazel run //bindings/pydrake:py/symbolic_test
> ```

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10412)
closed 2019-01-15 by EricCousineau-TRI