RobotLocomotion/drake #10413: geometry_set: Heterogeneous container is confusing, can lead to accidental loss of information

EricCousineau-TRI, 2019-01-15:
@avalenzu ran into an issue: He was trying to filter environment bodies + an articulated environment, and found that the collisions were still happening.

After debugging it for about 1.5 hours, he realized that it was because he was only adding *geometries* to the `GeometrySet`, not the *frames* as well.

The solution is fine (just call `Add(geometries, frames)`), but these pain points could be addressed.

After looking at the interface of `GeometrySet`, my observations of the interface:
* There are an absolutely *huge* number of overloads (with `enable_if`), for both constructors and `Add`.
* There is no `Add(GeometrySet)` overload; if this had been at the forefront, this wouldn't have happened.

My suggestion:
1. First add `Add(GeometrySet)`
2. Either: (a) Deprecate most, if not all, other overloads, and pay the cost of some malloc overhead, or (b) rename the granular `Add` methods, or mark them as advanced, etc., to try and dissuade users from using those, and instead use composition unless they *really* need the performance.

Perhaps the easiest solution for (2) is option (b).

@SeanCurtis-TRI Can I ask if you'd be fine with both (1) and (2)(b)?
I can also hand this over to you, but I think (1) and (2)(b) seem simple enough.

SeanCurtis-TRI, 2019-01-15:
I'm unconvinced. The statement "because he was only adding geometries to the GeometrySet, not the frames as well." Doesn't make sense to me. Adding frames is only a convenience overload -- a shorthand. It's a way to specify all of the geometries rigidly affixed to the frame without having to enumerate them. Strictly speaking it isn't necessary if you've already enumerated the geometries. 

In short, I don't understand either the problem nor the cause for confusion.  If the problem comes down to a misunderstanding of what's in the GeometrySet, then I feel the solution is in documentation and not eliminating very practical overloads.

It seems you've characterized the overload as a performance optimization and that's not remotely the case. So, clearly there is some disconnect between a) my intent, b) my implementation, c) my documentation, and d) others' understanding. Eliminating the API does *not* resolve that disconnect.

I feel the class documentaiton is pretty clear:

>The %GeometrySet, as its name implies, is a convenience class for defining a
 set of geometries. What makes it unique from a simple `std::set<GeometryId>`
 instance is that membership doesn't require explicit GeometryId enumeration;
 GeometryId values can be added to the set by adding the `FrameId` for the
 frame to which the geometries are rigidly affixed.

That said, I *love* the idea of adding an overload where the contents of one set can be added to another.

SeanCurtis-TRI, 2019-01-15:
This was a bit hard to grok, but I finally get it.

Certainly, this 100% supports the idea of being able to merge two geometry sets without worrying about the encoding at all.  (I.e., `GeometrySet::Add(const GeometrySet&)`).

Part of the problem is the fact that the API `GeometrySet::geometries()` and `GeometrySet::frames()` are not really intended to *be* public API; they are only there to facilitate testing. However, without any other mechanism to combine two sets, you dropped back and punted on whatever you could find.

So, definitely adding the new interface is definitely a big win. Arguably, the misleading APIs should be removed from public and the tests reformulated accordingly.

SeanCurtis-TRI, 2019-01-15:
In the PR, I have two options: kill the misleading API or deprecate the API.  Killing is simpler. I looked in anzu, and there are only usages of the `GeometrySet` in `dish_task_simulation.cc`. Obviously, this doesn't address any alternative branches that are out there.

Votes on deprecate vs kill outright?

EricCousineau-TRI, 2019-01-15:
I vote kill outright! I'd assume that no one else is using it too much right now.

SeanCurtis-TRI, 2019-01-15:
That's what I was hoping you'd say. I don't mind modding the PR to replace deprecation with killation, but am also willing to defer that. I strongly suspect that usages of this are practically nil.  Certainly the deprecated is untouched in Anzu.

PR #10417 (2019-01-22): [geometry] Clean up/extend GeometrySet API
This does two things in two commits:

1. Adds the interface for merging one `GeometrySet` into another.
2. Deprecates the misleading introspection methods so that it doesn't imply that users should try to poke in the internals of GeometrySet.

~relates to #10413~
EDIT (eric): Resolves #10413

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10417)
closed 2019-01-22 by SeanCurtis-TRI