RobotLocomotion/drake #10422: Finish the transition to the Fortran version of SNOPT

jwnimmer-tri, 2019-01-16:
The overall goal is to stop using f2c builds of SNOPT, instead _only_ using the Fortran builds.  We have many pieces of this already in place, but to finish it off we both need to add a few missing features or documentation, as well as resolve a few (possibly upstream) bugs.

This is just a placeholder for now.  I will update it with additional details a TODO lists, as we consolidate more details.

jwnimmer-tri, 2019-01-16:
Because we have (in TRI's internal applications) encountered problems with the newer SNOPT bindings, part of what we will need is to ask users to test the new SNOPT bindings in their own use-cases.  To do that, we'll need to ensure that most (or all) build or installation mechanisms have a path by which a user can opt-in to the new SNOPT.  Probably we should offer a CMake option for Fortran, as well as make a (possibly one-off manual) binary release that uses Fortran.

Similarly, we also need to make sure we have CI coverage for a reasonable subset of SNOPT-related build options.

I will add checklist items for these to the top post.  However, before we pursue those we should probably isolate and understand (and possibly resolve) the known shortcomings of the newer SNOPT bindings, since that might inform how we approach the build transition questions.

hongkai-dai, 2019-01-16:
> TBD Jeremy thinks that SNOPT 7.6 Fortran has new iteration-limit bugs on programs with costs?

That happened in Anzu, when we call the collision free planner. Snopt 7.6 fortran reached 1000 major iterations limit, without finding a solution. Snopt 7.4 fortran or Snopt 7.6 f2c don't have this problem.

EricCousineau-TRI, 2019-04-04:
Jamie has added Weekly builds here: https://drake-jenkins.csail.mit.edu/view/SNOPT%20Fortran/

EricCousineau-TRI, 2019-04-04:
@jwnimmer-tri Questions for ya per f2f with @jamiesnape:
* What job types do you want to spin up? Is Nightly or Continuous sufficient to tick the box?
   * Ideally, we don't add too many jobs before doing our flag day.
* Is there a reason to delay if all tests are passing on CI?
   * Jamie can't see where failing tests are disabled in the above checkbox. Is there something else to see?

jwnimmer-tri, 2019-04-11:
It's still on my TODO list.  I need to go back and update the plan, and discuss with the users of MathematicalProgram.  I am not convinced that a flag day is going to be practical, but will update here once I have useful information.

EricCousineau-TRI, 2019-05-16:
Removing `kitware` tag for the time being until the action is back on Kitware.

jwnimmer-tri, 2019-06-20:
> What job types do you want to spin up? Is Nightly or Continuous sufficient to tick the box?

I see a bunch of weekly builds for Snopt Fortran and Snopt F2C currently exist.  Those seem sufficient as far as job flavors for now.

jwnimmer-tri, 2019-06-20:
Instructions for downstream testing:
- For Bazel builds, see https://drake.mit.edu/bazel.html#test-the-build-for-either-mechanism for how to change.  Basically `--config snopt` is the default, but you can say `--config snopt_f2c` or `--config snopt_fortran` to force it.
- For CMake builds, you can change the options in the ccmake GUI.  If you currently set `WITH_SNOPT` to `=ON`, you can also say `=F2C` or `=Fortran` to force it.  Ditto for `WITH_ROBOTLOCOMOTION_SNOPT`.

jwnimmer-tri, 2019-08-01:
My plan is to pull the trigger on "Use gfortran by default" today or tomorrow.

jwnimmer-tri, 2019-08-05:
@hongkai-dai in the top post above, we have "Be clear about currently known bugs / performance degradations in SNOPT Fortran." with some TBD items under it.  I'm wondering if we can add more information to some of those known problems, yet?  Fortran is the default for our users, now.

hongkai-dai, 2019-08-05:
> TBD We will plan to have a collision free IK solver in Drake. We should see if SNOPT Fortan can solve it.

That collision free IK is in Drake, and I tested it in my local dev branch with snopt fortran. It works. I think we can remove this bullet point.

jwnimmer-tri, 2019-08-05:
Did you test with SNOPT/Fortran 7.6, or 7.4 though?  This issue is about enumerating failures of 7.6.  Our Drake default SNOPT is 7.4.

jwnimmer-tri, 2019-08-15:
To close the loop on this: from direct messages, @hongkai-dai says:
> ... [with SNOPT 7.6 the] result satisfy the collision avoidance constraint, but it generates different posture as with SNOPT 7.4.  The test I do is limited to a very simple ... [system]. For more complicated collision environment ... I haven't tested it.

I am still OK calling this "good enough" for now, so I'll leave that checkbox removed from the top post.

closed 2019-08-15 by jwnimmer-tri