RobotLocomotion/drake #10424: SNOPT DistanceToTetrahedron test failures

jwnimmer-tri, 2019-01-16:
New as of the SNOPT 7.6 release, when using the Fortran (not f2c) implementation, the solver reaches an iteration limit, on a program that it used to be able to solve.

---

Details and reproduction recipe:

When I run this case against various releases and modalities of SNOPT, then I see the following results:

**PASS**: `env SNOPT_PATH=snopt7-4.tar.gz bazel test -c dbg --config snopt_f2c --test_tag_filters= //solvers:snopt_solver_test --test_arg=--gtest_filter=SnoptTest.DistanceToTetrahedron`

**PASS**: `env SNOPT_PATH=snopt7-4.tar.gz bazel test -c dbg --config snopt_fortran --test_tag_filters= //solvers:snopt_solver_test --test_arg=--gtest_filter=SnoptTest.DistanceToTetrahedron`

**PASS**: `env SNOPT_PATH=snopt7.6.tar.gz bazel test -c dbg --config snopt_f2c --test_tag_filters= //solvers:snopt_solver_test --test_arg=--gtest_filter=SnoptTest.DistanceToTetrahedron`

**FAIL**: `env SNOPT_PATH=snopt7.6.tar.gz bazel test -c dbg --config snopt_fortran --test_tag_filters= //solvers:snopt_solver_test --test_arg=--gtest_filter=SnoptTest.DistanceToTetrahedron`

**PASS**: `env SNOPT_PATH=snopt7-4.tar.gz bazel test -c dbg --config snopt_f2c --test_tag_filters= //solvers:snopt_solver_test --test_arg=--gtest_filter=UnboundedLinearProgramTest0.TestSnopt`
```
solvers/test/snopt_solver_test.cc:173: Failure
Expected equality of these values:
  result
    Which is: IterationLimit
  SolutionResult::kSolutionFound
    Which is: SolutionFound
...
```

According to `README.WhatsNew`:
- My `snopt7-4.tar.gz` is `Version 7.4-1.1`.
- My `snopt7.6.tar.gz` is `Version 7.6.0`.

Drake was commit `62d46e7515bbf00a936bd634aeb11db204442805` for this report.

jwnimmer-tri, 2019-01-16:
I suspect that this bug is the same as (or closely related to) one of the bugs reported upstream in snopt/snopt-interface#19.  However, that issue is meandering and confusing, so it's difficult to say.  Perhaps @hongkai-dai can clarify.

@hongkai-dai To the extent we have an inventory of known problems related to iteration explosions in SNOPT 7.6, we should be sure to transcribe them into Drake issues (especially, with known-to-fail unit tests PR'd to master is even better).  That let's us (at a minimum) reproduce and test them using MathematicalProgram, even if that's not directly usable by upstream to debug and resolve our bug reports.

If we want to have separate Drake issues for iteration limits on feasibility problems vs problems with a cost, that's probably a good idea.  It sounds from the upstream report that the two program classes are handled differently.

This particular bug (DistanceToTetrahedron) looks to be a feasibility problem.  We can open an new Drake issue for programs with costs, perhaps?

hongkai-dai, 2019-01-16:
> I suspect that this bug is the same as (or closely related to) one of the bugs reported upstream in snopt/snopt-interface#19

Yes it is the same problem. This is a feasibility problem. The symptom is that the solver takes 1000 major iterations, with very few (13) minor iterations. That is not the normal behavior of an SQP solver. Normally it takes more minor iterations than major iterations.