RobotLocomotion/drake #10435: arm64 support? Bazel build fails on NVIDIA Jetson TX2

5yler, 2019-01-17:
Platform: out of the box NVIDIA Jetson TX2 with Ubuntu 16.04
```bash
$ uname -a
Linux tegra-ubuntu 4.4.38-tegra #1 SMP PREEMPT Fri Jul 28 09:55:22 PDT 2017 aarch64 aarch64 aarch64 GNU/Linux
```
I successfully installed prereqs on latest master, with the exception of bazel (the prereqs script tries to pull down an `amd64` binary, so I built 0.21.0 from source).

Output of `bazel build //...`:
```
INFO: Invocation ID: 4ec3a940-eb8c-407a-b952-65b3005b9f7b
INFO: Analysed 7811 targets (2 packages loaded, 5366 targets configured).
INFO: Found 7811 targets...
ERROR: /home/nvidia/.cache/bazel/_bazel_nvidia/20f9ec868d4333090f45a3fe92d28c3f/external/lcm/BUILD.bazel:144:1: Linking of rule '@lcm//:lcm-gen' failed (Exit 1) gcc failed: error executing command /usr/bin/gcc -o bazel-out/host/bin/external/lcm/lcm-gen '-Wl,-rpath,$ORIGIN/../../_solib_aarch64/_U@lcm_S_S_Clcm___Uexternal_Slcm' -Lbazel-out/host/bin/_solib_aarch64/_U@lcm_S_S_Clcm___Uexternal_Slcm ... (remaining 8 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
bazel-out/host/bin/_solib_aarch64/_U@lcm_S_S_Clcm___Uexternal_Slcm/libdrake_lcm.so: error: undefined reference to 'pthread_sigmask'
collect2: error: ld returned 1 exit status
INFO: Elapsed time: 1167.759s, Critical Path: 208.92s
INFO: 505 processes: 505 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
```
Not sure if this is a platform dependent issue, and curious if anyone has built successfully on `arm64`. I tried checking the existing issues and wasn't able to find any related discussion. Please advise.

jwnimmer-tri, 2019-01-17:
Our https://drake.mit.edu/developers.html#supported-configurations page indicates what we support, but I see now that it doesn't say which architectures.  Oops!  On Ubuntu, our CI only covers `amd64`.

While I'd hope that our code is portable, I think it's unlikely that all of `bazel build //...` would run to completion flawlessly on the first try.  I'd advise starting with a narrower subset like `bazel build //common` and grow from there -- only building the pieces you're going to want to use, such as `//multibody/plant` or whatever it is.

For the particular LCM error you posted, it looks like we failed to declare a pthread dependency somewhere?  You could try using `bazel build --copt=-pthread` flag to inject the `-pthread` flag to every C++ compilation step, at least for now.

I'll leave this issue open as a forum in case anyone else wants to chime in that they've gotten it working, or has related experiences, or wants to advocate for us adding support for the `arm64` architecture.

In the meantime, though, I'm not aware of anyone else having tried this yet, so you might be breaking new ground here.

jamiesnape, 2019-01-23:
> While I'd hope that our code is portable, I think it's unlikely that all of bazel build //... would run to completion flawlessly on the first try on arm64.

Since we use precompiled dependencies at various stages, we know with certainty that it will fail. At a minimum, you would need to avoid any target at depends on `@vtk` and any tests that use `@buildifier`. Also, you would not be able to use optional dependencies `@gurobi` and `@mosek`.

jwnimmer-tri, 2019-03-01:
For anyone coming here from a search engine, I'm happy to take more comments and suggestions, but for now (with the documentation updates noting x86_64), I'll close this out for now.

closed 2019-03-01 by jwnimmer-tri

jwnimmer-tri, 2020-06-08:
=> https://github.com/RobotLocomotion/drake/issues/13514 for additional discussions