RobotLocomotion/drake #10436: [geometry] GeometryState::X_WG_proximity_ never shrinks.

avalenzu, 2019-01-17:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
> * Have you reviewed existing GitHub issues and StackOverflow posts?
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

SeanCurtis-TRI, 2019-01-17:
I'm assuming this became manifest from a sequence of adding, removing, and then adding geometry?

avalenzu, 2019-01-17:
Yes. Although only the adding and removing steps are required to make it manifest.

PR #10441 (2019-01-18): Fix error in geometry removal
The mapping from proximity index to geometry index for dynamic geometries
was not being updated and leading to crashes when collision queries were
made after geometry removal.

- Add a test that exercises the failure point.
- Add some additional tests to look at more correlated results
- Fix the bug

fixes #10436

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10441)
closed 2019-01-18 by SeanCurtis-TRI