RobotLocomotion/drake #10445: Deprecate and replace DoPublish() and other event dispatchers

sherm1, 2019-01-18:
As a final nail in the coffin of the old method for defining event, and to catch any hidden dependencies on the simulator's default publish-per-step behavior that changed in #10428, do the following:
1. Deprecate the `LeafSystem` virtuals `DoPublish(), DoCalcDiscreteVariableUpdates(), and DoCalcUnrestrictedUpdate()`.
2. Find all overrides of those in Drake and Anzu and replace them with the Event/callback equivalents (the default implementations should quietly remain). 
3. After the deprecation period expires, replace these methods with status-returning virtual methods `DoDispatchPublishEvents()`, `DoDispatchDiscreteUpdateEvents()`, and `DoDispatchUnrestrictedUpdateEvents()`.
4. Documentation for the latter should say "(Advanced)" and discuss the (rare) cases in which one might want to overload dispatchers for special handling of simultaneous events.
5. Fix internal framework code to bubble up the event status returns.

See issue #4447, pr #9292, pr #10428.

jwnimmer-tri, 2019-01-18:
Instead of adding more `DoDispatchPublishEvents` virtuals, I'd advocate for having a `DeclareCustomPublishDispatcher(functor)` etc API instead.  I think we could do that even during / concurrently with the `DoPublish` deprecation window.

Like all of the other System facets, by using `Declare` we make it easier to overload / add arguments / deprecate the feature in the future.

sherm1, 2019-01-18:
Oh, yeah -- that's much nicer.

sherm1, 2019-01-18:
Per f2f w/@edrumwri first step will be to replace as many Drake DoPublish() overrides as we can with Event/callbacks.

sherm1, 2019-01-25:
#10458 took care of step 1 for C++ code. Next up is to deal with DoPublish in python wrappings.

jwnimmer-tri, 2024-11-01:
I believe #20094 and #20714 and similar have fixed this.

closed 2024-11-01 by jwnimmer-tri