RobotLocomotion/drake #10449: Storing velocity, acceleration and torque limits in Joint.

siyuanfeng-tri, 2019-01-18:
It would be much nicer to store those limits in the same place as the position limits. We have to pass them around in separate vectors along with the MBP at the moment. The current use case is just passing those limits to some planning code, so I am totally happy to not enforce them in the simulation yet. I am also fine without parsing support initially. 
What do @amcastro-tri @sherm1 think? If you guys have no objections, I can make a draft PR. 

cc @avalenzu @calderpg-tri @hongkai-dai @EricCousineau-TRI

amcastro-tri, 2019-01-18:
I think your proposal does make sense.

sherm1, 2019-01-18:
I like the idea of storing more joint-related info with the joint. Should be very clear that we are just _storing_ these limits, though, not enforcing them. We should consider offering some enforcement at some point but for the simulator that means we have to model a physical mechanism for doing so (force-producing constraint, spring/damper system, or whatever).

EricCousineau-TRI, 2019-01-18:
+1 on storing 'em in the joints; we should still expose some succinct accessors, like @mposa's mods for `GetPositionLower/UpperLimtis()`.

sammy-tri, 2019-01-22:
Assigning to @sherm1 as default owner for dynamics issues.  Feel free to reassign.

sherm1, 2019-01-22:
Reassigned to @siyuanfeng-tri since he's already addressing this in #10454.

siyuanfeng-tri, 2019-01-23:
closed by #10454

closed 2019-01-23 by siyuanfeng-tri