RobotLocomotion/drake #10451: py autodfif: `np.linalg.norm` may not propagate gradients correctly?

EricCousineau-TRI, 2019-01-18:
@manuelli ran into an issue where formulating a cost in MP using Python caused a program to fail, even if he ended up multiplying the resulting expressing by zero:
```
weight = 0.  # need to confirm

def my_cost_bad(x):
    return weight * np.linalg.norm(x)**2

def my_cost_good(q):
    return weight * np.dot(x, x)
```

This may be due to gradients not propagating correctly.

EricCousineau-TRI, 2019-01-18:
Found the minimal repro:
```py
import warnings
import numpy as np
from pydrake.all import AutoDiffXd

warnings.simplefilter("error", Warning)
x = AutoDiffXd(0)
print(x.sqrt())  # Works
print(np.sqrt(x))  # Fails with divide-by-zero error
```

I have no idea why the first works, but the second fails...
@eric-wieser If you have a chance, can I ask if you might know why?

rpoyner-tri, 2026-11-12:
Closing as a duplicate of #17898. This is the same underlying problem: NumPy floating-point warnings (here, divide-by-zero from `np.sqrt` on `AutoDiffXd`) fire when NumPy operates on Drake's object-dtype scalars. Drake hides them with an import-time `ignore` filter, but they reappear if the user's own warning filter is applied after `import pydrake`. I confirmed this still happens with the Drake 1.58.0 wheel. #17898 stays open as the tracker.

closed 2026-11-12 by rpoyner-tri