RobotLocomotion/drake #10482: Insufficient geometry introspection for visualization

calderpg-tri, 2019-01-23:
Available geometry introspection allows the `Shape` of the geometry to be retrieved, but does not allow the role, source, or properties of the geometry to be retrieved. In the context of an export-to-RViz visualizer, the following need to be available:

- Source name for a given `GeometryId` or `FrameId`. This could be accomplished by making `GeometryState::get_source_id(FrameId frame_id)` public.
- `Role` of geometry, to distinguish between visual and collision geometry.
- `IllustrationProperties` of visual geometry.

The existing export-to-LCM pipeline relies on a number of internal methods to retrieve this information, so the simplest test that sufficient information is exposed is for `GeometryVisualizationImpl::BuildLoadMessage` to only use public methods and types.

@SeanCurtis-TRI I've assigned you for now since you did the other geometry introspection work.

SeanCurtis-TRI, 2019-01-23:
Several thoughts:

  - `get_source_id()` will *never* be public. `SceneGraph` cannot leak `SourceId`s otherwise one system can unilaterally hijack another systems "property". That's not to say that a `SourceId` can't be shared, but `SceneGraph` cannot be the agent of that. However, `SceneGraphInspector::get_source_name()` seems reasonable.
  - The `SceneGraphInspector` *does* have [`GetProximityProperties()`](https://github.com/RobotLocomotion/drake/blob/master/geometry/scene_graph_inspector.h#L300) and [`GetIllustrationProperties()`](https://github.com/RobotLocomotion/drake/blob/master/geometry/scene_graph_inspector.h#L312). (The latter directly addresses your third point). It also implicitly reports on the role -- if either returns `nullptr` the the geometry does not have the corresponding role.

So, as I see it, what you're missing is access to the name of the source that owns an entity.

Something you should know, this quarter I'm going to add the generic visualization interface. It includes two ports: one for getting a description of the scene sufficient for populating a visualizer (which should be evaluated in an "initialization" sense) and the other port that reports poses for all of the frames indicated in the former. So, all of these other things are stop-gap efforts.

sherm1, 2019-01-23:
Those are some good arguments, @calderpg-tri !

SeanCurtis-TRI, 2019-01-23:
- `SourceId`s are only necessary to change things in `SceneGraph`. They are unnecessary for learning about it (i.e., I can look up the properties of an arbitrary GeometryId, whether I have the `SourceId` or not, but I can't remove it, or hang a geometry on someone else's frame, etc.)
- I'm unclear on your second point. It seems to touch on two subjects:
  - Division between `GeometryState` and `SceneGraphInspector`. 
    - If there appears a conflict in information, it's because you're looking into a non-public API. The GeometryState is not for public consumption (well, not anymore). It would be clearer if I moved it into an `internal` namespace (but that's one of the *technical debt* issues that I haven't tackled because I have so much *new feature* debt. The division is simple: `SceneGraph` is used to populate the world, `QueryObject` is used to perform queries on the scene graph (note lower case) that depend on geometry source-calculated poses, and `SceneGraphInspector` provides queries that *don't* depend on those input poses. The fact that the *implementation* of all of those things happen to be wrapped into a single class should be an under-the-hood implementation detail.
    - Any code that is currently accessing the `GeometryState` directly *is* going to break in the "near" future. The current nature of `GeometryState` is a relic of a deprecated design goal and predates caching. To make `SceneGraph` cache compatible, `GeometryState` is going to change significantly.  Subverting the `SceneGraphInspector` and `QueryObject` is loading up on technical debt that will burn you in the future. Although, I am sympathetic with the gripe about evaluating contexts when you're not entangled with the Systems framework. A
  - The second issue isn't clear to me. I'm not sure what "keeping it private in the state makes no sense" means -- particular what the "it" in that sentence is. I assume it's related to the `GeometryState::get_source_id(FrameId)` method, but am otherwise unsure. Feel free to elaborate if you feel I'm missing something important.
- The visualization interface was actually a request of @RussTedrake (almost a year ago in #8473)  based on servicing a meshcat visualization. So, waiting on RViz isn't strictly necessary. 
  - However, your concern about being the *right* interface is dead on. However, It may turn out that simply using the `SceneGraphInspector` is enough. Although, I can conceive of something that walks the tree *for* you and filters out anything non-visualization would improve the experience. Right now, my primary concern about doing visualization is that you have to infer the structure of the tree encoded in the `SceneGraph` and that seems like an undue burden on the visualization code.

SeanCurtis-TRI, 2019-01-24:
I think you're biased by MBP's relationship with SG. MBP chose to register all its frames as children of the world frame: so `X_WF` frame is sufficient for describing the pose of every frame owned by MBP. But that's not a fundamental property of `SceneGraph`. `SceneGraph` can support arbitrary trees of frames (with geometries hanging on them). And to visualize that would require either a) rebuilding the tree in visualization, or b) re-calculating the X_WF from the various X_PF that are reported. Visualizers need to be expressed to be compatible with either.

I would offer a slight modification to the phrase "what are the geometries", to "what's in the scene and how is it organized?"  But I agree, that's "all" a visualizer needs to know.

tri-ltyyu, 2019-01-25:
Need to discuss the right code patterns that works for the different paths (based on visualization and non-visualizaiton uses). Next step is for Calder, Sean, and Eric should meet to talk about this.

tri-ltyyu, 2019-02-01:
Sean has created some test code and will review with Calder and team.

EricCousineau-TRI, 2020-04-28:
Per comment here: https://github.com/RobotLocomotion/drake/issues/9500#issuecomment-620722987

The introspection is actually quite nice, and *almost* accessible via `pydrake`. There are some items to be filled out in #13160 (which I hope to do sometime soon).

I've updated the overview to add a checklist item to update the existing usages to use `QueryObject` a bit more aggressively, because it is **wayyyyyyy** nicer than the hacky `BuildLoadMessage` + `PoseBundle` vestiges.

Thanks @SeanCurtis-TRI!!!

EricCousineau-TRI, 2020-04-28:
@SeanCurtis-TRI If you have a chance, could you critique the current usages of the `SceneGraph` introspection API to see if there are other improvements that I'm missing? You can find most of my hacks by searching for `TODO(` here:
https://github.com/EricCousineau-TRI/repro/tree/a8315dd0c8d1ee16ccda8747b6a32dcc969bc133/ros/drake_ros1_hacks

SeanCurtis-TRI, 2020-04-28:
I'll take a look.

SeanCurtis-TRI, 2020-04-28:
What's the best way to share thoughts?

EricCousineau-TRI, 2020-04-28:
I can make a PR on the current code with minor changes, so you can view the files. Will add link here.

EricCousineau-TRI, 2020-04-28:
Discussion PR: https://github.com/EricCousineau-TRI/repro/pull/1

EricCousineau-TRI, 2020-08-26:
See example of an Rviz-based visualizer using `QueryObject`:
https://github.com/EricCousineau-TRI/repro/blob/7a08250e55899dd47fb77ac6524331cd24d86e59/ros/drake_ros1_hacks/rviz_visualizer.py#L91-L96

FYI @pangtao22

EricCousineau-TRI, 2020-10-27:
Note that @SeanCurtis-TRI has a rendition of `QueryObject`-based Drake Visualizer in #14247

PR #15534 (2021-08-06): Deprecating PoseBundle
This deprecates `PoseBundle` as well as closely affiliated classes: `PoseVector`, `PoseAggregator`, `PoseBundoeToDrawMessage`.

There are two systems that have `PoseBundle`-valued output ports and two diagrams that export those ports.

The systems had explicit getters for the ports; those getters have been deprecated.

The diagrams were mixed. One had a getter, one relied on `get_output_port(name)`. To generally aid the case where a leaf output port is exported, the calc methods of the two systems have been given a "log once" warning.

Bindings and unit tests have likewi [... cut]
closed 2021-08-06 by jwnimmer-tri