RobotLocomotion/drake #10486: Installed (and binary release) drake-visualizer lacks useful default plugins

jwnimmer-tri, 2019-01-24:
When we run `bazel-bin/tools/drake_visualizer`, useful plugins like `show_image` are on by default.  When a user is running from an installed copy of Drake, they are off by default (and not even in the menus as an option, IIRC).  We shouldn't have to ask a user "_Which_ visualizer copy were you running" -- they should all just work the same.  Call to action: bake in the plugins to our installed drake-visualizer, as well.

\CC @EricCousineau-TRI @jamiesnape

jwnimmer-tri, 2019-01-24:
We could upstream the code into director, or we could install the director binary as `bin/drake-visualizer.real` and install our own `bin/drake-visualizer` tiny shim, or ...

jamiesnape, 2019-01-24:
So in effect #9653 only fixed half the problem of #9608?

EricCousineau-TRI, 2019-01-24:
Yup! The only pain point that I was aware of (but I see I did not explicitly state) was Drake + Anzu workflows, which was Bazel-centric.

This also partially relates the discussion for #7050 - e.g. how much mileage we'd get out of simple Python imports running `director --help` with `--script` specified ().

The scripts have been relatively stable, so I'd be fine with upstreaming them; however, I think just having them installed would be *kinda* easier. Just need to install to `drake` instead of `pydrake` (and ensure `__init__.py` modules are added), and then things should work just fine.

jwnimmer-tri, 2019-01-24:
Having an install-based workflow (instead of upstreaming the plugins) makes it easy-by-way-of-example for us to add more default plugins as we invent them.

jwnimmer-tri, 2020-04-23:
Here's the best workaround I've found so far.

This example uses the [Drake v0.17.0 binary release](https://github.com/RobotLocomotion/drake/releases/tag/v0.17.0) image for Ubuntu.  The CMake install should produce an identical result.

We need to use a source tree for the plugins, because the install tree does not have the plugins available anywhere.  But the python path doesn't work easily mixing the source and install trees, so we need to comment out the scoped_singleton_func stuff.

(1) Download and untar the image into `~/Downloads/v0.17.0`.

(2) Clone Drake git into `~/jwnimmer-tri/drake`.

(3) Patch each plugin like so:
```patch
--- a/tools/workspace/drake_visualizer/plugin/show_time.py
+++ b/tools/workspace/drake_visualizer/plugin/show_time.py
@@ -5,7 +5,7 @@ from director import lcmUtils
 from director import applogic
 from director.visualization import updateText
 
-from drake.tools.workspace.drake_visualizer.plugin import scoped_singleton_func
+# from drake.tools.workspace.drake_visualizer.plugin import scoped_singleton_func
 
 
 class TimeVisualizer(object):
@@ -78,7 +78,7 @@ class TimeVisualizer(object):
         updateText(my_text, 'text')
 
 
-@scoped_singleton_func
+# @scoped_singleton_func
 def init_visualizer():
     time_viz = TimeVisualizer()
     # Adds to the "Tools" menu.
```

(4) Run `~/Downloads/v0.17.0/drake/bin/drake-visualizer --script ~/jwnimmer-tri/drake/tools/workspace/drake_visualizer/plugin/show_time.py`

Bottom line, we really need to fix the install.

mjm522, 2020-04-23:
Using the info given by @jwnimmer-tri , I copied the plugin folder to the bin and it works. The repo is [here](https://github.com/mjm522/drake-binary-with-plugins). This is a quick fix for anyone using the binary to get all the plugins activated.

jamiesnape, 2020-04-23:
It's not a huge change, just a different `PYTHONPATH` ultimately.

EricCousineau-TRI, 2020-04-23:
Took longer, but I think I also ended up solving some form of #7050.

WIP Branch: https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue-10486-wip

PR #13127 (2020-04-29): drake_visualizer: Install builtin scripts
Put under test, both in Bazel and install

Resolves #10486 
Resolves #7050.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13127)
closed 2020-04-29 by EricCousineau-TRI