RobotLocomotion/drake #10487: missing / misleading bindings for publishing rgbd cameras from pydrake

RussTedrake, 2019-01-24:
I'm trying to reproduce the following lines
https://github.com/RobotLocomotion/drake/blob/master/examples/manipulation_station/proof_of_life.cc#L48-L65
in `end_effector_teleop.py`

We don't currently have bindings for `systems::sensors::ImageToLcmImageArrayT`, and my quick attempt was stymied by the complexity of dealing with the pixeltype templates.

I'm afraid I also struggled a bit to use LcmPublisherSystem:
```
>>> from robotlocomotion import image_array_t
>>> from pydrake.systems.lcm import LcmPublisherSystem
>>> a = LcmPublisherSystem.Make("test",image_array_t(), None, 0.1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: _make_lcm_system() takes exactly 4 arguments (5 given)
```

@EricCousineau-TRI -- might you be able to help?  this a step in the direction of moving manipulation station to be a thin python-only example.  ;-)

EricCousineau-TRI, 2019-01-24:
Sounds good! Will take a 30min timeboxed try later today, and then consider enlisting help if it takes longer.

EricCousineau-TRI, 2019-01-25:
Ah, on the error message you received, two things:
* It was originally wrong because you passed an object instantiation (`image_array_t()`) rather than the type (`image_array_t`).
* The `@classmethod` was changed to `@staticmethod` in #10455 (as a follow-up to the change added in #10341).

So I don't think there's anything to really change; will still try to rename the method for a bit more of user-friendliness.

closed 2019-01-28 by EricCousineau-TRI