RobotLocomotion/drake #10493: MBP needs templated versions of get_joint and get_mutable_joint

RussTedrake, 2019-01-24:
that match the templates on `GetJointByName`, e.g.
```
  template <template <typename> class JointType = Joint>
  const JointType<T>& get_joint(JointIndex joint_index) const {
```
I'm finding myself down-casting downstream, which isn't the end of the world but is inconsistent with the `GetJointByName` workflow.

amcastro-tri, 2019-01-29:
@EricCousineau-TRI, do you like these templated methods? I remember that you probably didn't quite like them? do I do see the value of having this sugar.

EricCousineau-TRI, 2019-01-29:
I definitely see value in C++ land when dynamic polymorphism + downcasting is involved.

The thing I don't like is the only implementation being so tightly coupled to the templated method that it's hard to figure out what's sub-type specific, and what is generic to the base type.

In this case, I would suggest something like:
```c++
// Put the real code here.
const Base& GetSomething(Index);

// Pure sugar.
template <typename Type>
const Type& GetSomething(Index) const {
  return dynamic_cast<const Type&>(GetSomething(index));
}
```

... if it doesn't cause ambiguity issues.

jwnimmer-tri, 2019-01-29:
Consider `GetSomething(index)` and `GetSomethingAs<Subtype>(index)` to disambiguate the base vs downcast type.

amcastro-tri, 2019-01-29:
Thank you for the input guys, greatly appreciated.

jwnimmer-tri, 2024-11-01:
Closing as not planned.  Python is the way and the light, and this is C++-only sugar.

closed 2024-11-01 by jwnimmer-tri