RobotLocomotion/drake #10517: MultibodyPlant does not provide an option to feed a single actuation input port, when multiple model instances exist

rcory, 2019-01-28:
Users should be able to provide a single actuation vector into one MBP input port, even in cases where multiple model instances exists. The current implementations of `MultibodyPlant::get_actuation_input_port` provide a single input port _per model instance_. 

This would mirror the functionality provided by `MultibodyPlant::get_continous_state_output_port()`, which is overloaded to produce the full state vector when multiple model instances exists.

sherm1, 2019-01-29:
This seems like a great idea -- the input port and output port would match, and be compatible with the ID controller.

edrumwri, 2019-01-29:
Rick and I have been discussing and I offered to help with a first iteration. @amcastro-tri, what do you think of me altering the get_actuation_input_port() method - i.e., the one that _currently_ throws an exception if there is more than one model instance- to do this instead? Backward compatibility will be maintained, but there will be a semantic change (one that introduces new capabilities, which smells kosher to me).

sherm1, 2019-01-30:
Marking this high pri since it is tormenting Rick.

rcory, 2019-02-22:
PR #10614 resolves this issue.

closed 2019-02-22 by rcory