RobotLocomotion/drake #10526: meshcat: Deprecation warning

EricCousineau-TRI, 2019-01-29:
From this test (the top-level bit):
https://drake-cdash.csail.mit.edu/testDetails.php?test=25858064&build=1110775

```
meshcat_python/src/meshcat/transformations.py:1042: DeprecationWarning: invalid escape sequence \*
  """
```
Reproducible via:
```
$ bazel run --config=python3 //bindings/pydrake:py/all_test
```

jamiesnape, 2019-01-29:
https://github.com/rdeits/meshcat-python/blob/d0c8b6a9d1d750495ef9513254761cc14773cf99/src/meshcat/transformations.py#L1001

Can you file an upstream bug?

EricCousineau-TRI, 2019-02-05:
@jamiesnape Now that this is updated, should we update the fork?

jamiesnape, 2019-02-05:
Indeed. There are a whole host of other SHA bumps in PR at the moment, so makes sense.

jamiesnape, 2019-02-05:
You may want to catch #10575 before it merges.

jwnimmer-tri, 2019-02-05:
I considered that, but it's already passed all CI.  Since you'll need a macOS CI spin anyway for this bump, we might as well do that in a new PR.

jwnimmer-tri, 2019-02-05:
(I could still roll it into #10576 or #10578 though, if you want to wait on those being ready.)

jamiesnape, 2019-02-05:
I think a separate PR is fine. I though #10575 was updating it already to a different SHA, but that was `meshcat`, not `meshcat_python`.

closed 2019-02-07 by EricCousineau-TRI