RobotLocomotion/drake #10527: py3: `ImportWarning: can't resolve package from __spec__ or __package__, falling back ...`

EricCousineau-TRI, 2019-01-29:
From this test:
https://drake-cdash.csail.mit.edu/testDetails.php?test=25858064&build=1110775

Reproducible via:
```
$ bazel run --config=python3 //bindings/pydrake:py/all_test
```

EricCousineau-TRI, 2019-01-29:
Seems like a `meshcat` + `zmq` issue? Passing the buck on to you, @jamiesnape.

With this patch:
```patch
diff --git a/bindings/pydrake/test/all_test.py b/bindings/pydrake/test/all_test.py
index d40071141..ad910b92a 100644
--- a/bindings/pydrake/test/all_test.py
+++ b/bindings/pydrake/test/all_test.py
@@ -4,6 +4,11 @@ import sys
 import unittest
 import warnings
 
+warnings.filterwarnings("ignore", category=DeprecationWarning, message="invalid escape")
+warnings.filterwarnings("error", category=ImportWarning)
+import pydrake.all
+exit(1)
+
 
 class TestAll(unittest.TestCase):
     # N.B. Synchronize code snippests with `doc/python_bindings.rst`.
```

Part of the stacktrace:
```
Traceback (most recent call last):
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/drake/common/test_utilities/drake_py_unittest_main.py", line 63, in <module>
    module = imp.load_source(test_name, found_filename)
  File "/usr/lib/python3.6/imp.py", line 172, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 684, in _load
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "./bindings/pydrake/test/all_test.py", line 9, in <module>
    import pydrake.all
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/drake/bindings/pydrake/all.py", line 50, in <module>
    from .systems.all import *
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/drake/bindings/pydrake/systems/all.py", line 5, in <module>
    from .meshcat_visualizer import *
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/drake/bindings/pydrake/systems/meshcat_visualizer.py", line 11, in <module>
    import meshcat
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/meshcat_python/src/meshcat/__init__.py", line 5, in <module>
    from . import visualizer
  File "/.../bin/bindings/pydrake/py/all_test.runfiles/meshcat_python/src/meshcat/visualizer.py", line 10, in <module>
    import zmq
  File "/usr/lib/python3/dist-packages/zmq/__init__.py", line 34, in <module>
    from zmq import backend
  File "/usr/lib/python3/dist-packages/zmq/backend/__init__.py", line 40, in <module>
    reraise(*exc_info)
  File "/usr/lib/python3/dist-packages/zmq/utils/sixcerpt.py", line 34, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/zmq/backend/__init__.py", line 27, in <module>
    _ns = select_backend(first)
  File "/usr/lib/python3/dist-packages/zmq/backend/select.py", line 33, in select_backend
    reraise(ImportError, ImportError("Importing %s failed with %s" % (name, e)), exc_info[2])
  File "/usr/lib/python3/dist-packages/zmq/utils/sixcerpt.py", line 33, in reraise
    raise value.with_traceback(tb)
  File "/usr/lib/python3/dist-packages/zmq/backend/select.py", line 26, in select_backend
    mod = __import__(name, fromlist=public_api)
  File "/usr/lib/python3/dist-packages/zmq/backend/cython/__init__.py", line 6, in <module>
    from . import (constants, error, message, context,
  File "zmq/backend/cython/error.pyx", line 32, in init zmq.backend.cython.error (zmq/backend/cython/error.c:1237)
ImportError: Importing zmq.backend.cython failed with can't resolve package from __spec__ or __package__, falling back on __name__ and __path__
```

If it's due to just `cython`, we can always scrub the warning for now.

jamiesnape, 2019-01-30:
Since we use the system `zmq` this probably is not going to get fixed anytime soon.

PR #10536 (2019-01-30): py meshcat_visualizer: Ignore import warnings in Python 3
Resolves #10527 (as a workaround)

This can be benefited by https://github.com/PyCQA/pycodestyle/pull/834 when it comes to filtering import warnings.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10536)
closed 2019-01-30 by EricCousineau-TRI