RobotLocomotion/drake #10542: CMake support for snopt_fortran transition

jwnimmer-tri, 2019-01-30:
Per the meta-issue #10422 for the snopt_fortran transition, we need to: "Document how to test-drive `--config snopt_fortran`, covering all build/installation modalities."

Part of that is adding the CMake flavor of that support and documentation:
(1) CMake from-source builds should offer the choice (f2c or fortran) to the user.
(2) Nightly releases should offer both flavors (during the transition period).

For (1), in the `drake/CMakeLists.txt` there should be an option that lets the user control whether they get `--config snopt_f2c` or `--config snopt_fortran` in their generated `cmake/bazel.rc` file.  At some point, we would drop f2c support and remove this option.  The default should be the same as the bazel `--config snopt` (so, f2c for now).  At some point soon, we would flip the default to be fortran for both CMake and Bazel builds.

For (2) what we can't do is stop providing f2c in the nightly releases.  I think a fair compromise is to wait until we switch our default to fortran, and at that point add a nightly-f2c release image for a transition period (we could spin up that build a few days early, even though it would be a bit redundant).

Does this sound sensible?

jwnimmer-tri, 2019-01-30:
@jamiesnape Can I hand this issue over to you, to do the PRs?

\CC @EricCousineau-TRI

jamiesnape, 2019-01-30:
Yes, that's fine.

jwnimmer-tri, 2019-02-12:
@jamiesnape For #10422, soon I would like to start asking users to test-drive the Fortran build of snopt.  Could I advocate for step (1) above (adding a choice to CMakeLists) to happen in the next week or so?

jamiesnape, 2019-02-12:
Yes.

EricCousineau-TRI, 2019-03-14:
Is the `high` priority still relevant?

Can I ask if there is someone heavily invested in this solution so that they can review using the releases and provide feedback on an `experimental` PR with a parameterized build so that we can test this specific release?

jwnimmer-tri, 2019-03-14:
Yes, the fortran transition in general is high priority, so any issue that is still relevant / open is high priority.

I do not understand at all what you are asking in the second question.  I am happy to test any prototypes, if that's what you're asking?

EricCousineau-TRI, 2019-03-14:
SGTM.

And yup! Ultimately asking for someone who'll review the solution.

EricCousineau-TRI, 2019-03-28:
### Part A
Per f2f, the first checkboxes on #10422 (fix failing tests) seems like a prerequisite to have passing CI. Should that be resolved first before we invest effort into releasing packages? Or do we need special tags?

I've added a checkbox above to have CI for Snopt fortran.

### Part B
Also, providing a multiple release types requires #10948, which would require 20 PEHs of Jamie's time; at present, our downloads for released packages are still relatively small.

Do we still need binary releases to be part of high priority, or can that be sloughed onto the back burner? Can we just make the binary release switch a flag day?
(That being said, #10804 would benefit, but it's lower priority pending legal bits.)

(I would like Python improvements to take priority over this if it can be placed on the backburner.)

jwnimmer-tri, 2019-03-28:
> Per f2f, the first checkboxes on #10422 (fix failing tests) seems like a prerequisite to have passing CI. Should that be resolved first before we invest effort into releasing packages? Or do we need special tags?

Those tests are marked expected-fail / ignored as necessary.  They are not blocking this issue, or the transition.

jwnimmer-tri, 2019-03-28:
> I've added a checkbox above to have CI for Snopt fortran.

I have removed that checkbox from the top post; I would prefer to track it with the checkbox in #10878, since it's relevant to both Bazel and CMake builds, but most relevant to the Bazel builds (which actually run the tests).

jwnimmer-tri, 2019-03-28:
Since (2) proposes that it can be deferred until we start the transition period ("a fair compromise is to wait until we switch our default to gfortran"), and that period is not yet upon us, I'll lower the priority here for now.  But if indeed it is such a chore to refactor the CI, it might still be worth getting underway, so that it's ready when we need it (in 3? 6? weeks from now).

EricCousineau-TRI, 2019-03-28:
Er, is #10878 the right issue? Or did you mean #10422?

jamiesnape, 2019-03-28:
> our downloads for released packages are still relatively small.

BTW We are talking single digits (excluding `drake-external-examples`). It is plausible we may get ~0 downloads during a transition period. Hence why the issue asks whether the refactoring is necessary.

jwnimmer-tri, 2019-03-28:
There is every chance that the transition period is unboundedly long (like, months, or a year, or ...).

jamiesnape, 2019-03-28:
(An unboundedly long transition period opens up a whole host of awkward questions that probably move to a different issue.)

jamiesnape, 2019-03-28:
My argument is that the transition should be complete before you start offering binary packages. That there is a (long) transition period in the first place suggests a less than 100% confidence in the new configuration. But all means have a transition for source builds, but the binary packages should be the definitive version of Drake.

jwnimmer-tri, 2019-06-20:
> ... the binary packages should be the definitive version of Drake.

I'll buy this.  And so, there's nothing left to do in this issue.

closed 2019-06-20 by jwnimmer-tri