RobotLocomotion/drake #10546: Actuation coordinate ordering does not necessarily match ordering of velocity coordinates in MBP

edrumwri, 2019-01-30:
The following discussion assumes that the number of actuation variables equals the number of velocity variables in MBP. 

It is the case that the ordering of actuation variables may not match the ordering of velocity variables, _at least (but not necessarily limited to) the case where the number of model instances is greater than one_.

Rick discovered this counterintuitive behavior when attempting to use the inverse dynamics controller to control a simple plant comprised of three model instances (his code provided inline below). When connecting the output of the ID controller to a feature-in-progress that adds a single actuation port (for controlling all model instances) to MBP, the controls do not match those obtained by demultiplexing the ID control output and feeding the results into the model-instance-actuation ports of the MBP.

I've been able to work around this problem to keep Rick going, but only with an ugly kludge to the inverse dynamics controller. Observe the necessary changes to inverse_dynamics.cc starting at Line 82:

    const VectorX<T> output_gv = plant.CalcInverseDynamics(
        *multibody_plant_context_, desired_vd, external_forces);
    Eigen::VectorBlock<VectorX<T>> output_u = output->get_mutable_value();
    for (multibody::ModelInstanceIndex i(0); i < plant.num_model_instances(); ++i) {
      plant.SetActuationInArray(i, plant.GetVelocitiesFromArray(i, output_gv),
                                &output_u);
    }

Previously, we only had to do:

    output->get_mutable_value() = plant.CalcInverseDynamics(*multibody_plant_context_, desired_vd, external_forces);

We'd like to PR the input actuation port (Rick and I working together), but the simple test that Rick devised breaks if I don't introduce this ugly kludge to inverse dynamics, assuming that it will take Alejandro a while to correct this architecture problem. Question for discussion: shall we submit the PR anyway? Inverse dynamics is kind of broken without it...

Rick's simulation code:
```
"""Provides an example of a 2d gripper, with prismatic fingertips."""

from pydrake.common import FindResourceOrThrow
from pydrake.geometry import (ConnectDrakeVisualizer, SceneGraph)
from pydrake.lcm import DrakeLcm
from pydrake.systems.lcm import LcmPublisherSystem
from pydrake.multibody.multibody_tree import UniformGravityFieldElement
from pydrake.multibody.multibody_tree.multibody_plant import MultibodyPlant
from pydrake.multibody.parsing import Parser
from pydrake.systems.framework import DiagramBuilder
from pydrake.systems.analysis import Simulator
from pydrake.math import RigidTransform, RollPitchYaw, RotationMatrix

from pydrake.systems.controllers import InverseDynamicsController

from pydrake.systems.primitives import ConstantVectorSource

import numpy as np

def add_gripper_model(gripper, file_name):
    # Weld the first finger
    finger1 = Parser(plant=gripper).AddModelFromFile(file_name, "finger1")
    X_PC1 = RigidTransform(RollPitchYaw([0, 0, 0]), [0, 0, 0.18])
    child_frame = gripper.GetFrameByName("ground", finger1)
    gripper.WeldFrames(gripper.world_frame(),
                       child_frame, X_PC1.GetAsIsometry3())

    # Weld the 2nd finger
    finger2 = Parser(plant=gripper).AddModelFromFile(file_name, "finger2")
    X_PC2 = RigidTransform(
        RollPitchYaw([120*(np.pi/180.), 0, 0]), [0, 0, 0]).multiply(X_PC1)
    child_frame = gripper.GetFrameByName("ground", finger2)
    gripper.WeldFrames(gripper.world_frame(),
                       child_frame, X_PC2.GetAsIsometry3())

    return finger1, finger2

def main():
    file_name = FindResourceOrThrow(
        "drake/examples/2d_gripper/2d_prismatic_finger.sdf")
    builder = DiagramBuilder()
    scene_graph = builder.AddSystem(SceneGraph())
    gripper = builder.AddSystem(MultibodyPlant(0.001))
    gripper.RegisterAsSourceForSceneGraph(scene_graph)

    # Add the fingers to the simulation plant.
    (finger1_id, finger2_id) = add_gripper_model(gripper, file_name)

    gripper.AddForceElement(UniformGravityFieldElement())

    # Finalize the plants.
    gripper.Finalize()
    assert gripper.geometry_source_is_registered()

    # Connect the SceneGraph with MBP.
    builder.Connect(
        scene_graph.get_query_output_port(),
        gripper.get_geometry_query_input_port())
    builder.Connect(
        gripper.get_geometry_poses_output_port(),
        scene_graph.get_source_pose_port(gripper.get_source_id()))

    # Add an ID controller to hold the fingers in place.
    Kp = np.array([1, 1, 1, 1]) * 50
    Kd = np.array([1, 1, 1, 1]) * 10
    Ki = np.array([1, 1, 1, 1]) * 10
    id_controller = builder.AddSystem(
        InverseDynamicsController(gripper, Kp, Ki, Kd, False))

    # Connect the ID controller
    builder.Connect(gripper.get_continuous_state_output_port(),
                    id_controller.get_input_port_estimated_state())

    desired_state = [0, 0, 0, 0, 0, 0, 0, 0]

    # Connect the desired state
    const_src = builder.AddSystem(ConstantVectorSource(desired_state))
    builder.Connect(const_src.get_output_port(0),
                    id_controller.get_input_port_desired_state())

    # Connect the single actuation port
    builder.Connect(id_controller.get_output_port_control(),
                    gripper.get_actuation_input_port())

    ConnectDrakeVisualizer(builder=builder, scene_graph=scene_graph)

    diagram = builder.Build()

    diagram_context = diagram.CreateDefaultContext()
    gripper_context = diagram.GetMutableSubsystemContext(
        gripper, diagram_context)

    # Fix the model instance actuation ports to zero. These are added to the
    # single vector actuation input port returned by get_actuation_input_port().
    # Produces errors if these ports are left unconnected.
    gripper_context.FixInputPort(
        gripper.get_actuation_input_port(finger1_id).get_index(), [0, 0])
    gripper_context.FixInputPort(
        gripper.get_actuation_input_port(finger2_id).get_index(), [0, 0])

    simulator = Simulator(diagram, diagram_context)
    simulator.set_publish_every_time_step(False)
    simulator.set_target_realtime_rate(1.0)
    simulator.Initialize()

    while simulator.get_context().get_time() < 3.0:
        simulator.StepTo(simulator.get_context().get_time() + 0.01)

if __name__ == "__main__":
    main()  # This is what you would have, but the following is useful.
```

cc @rcory @sherm1

sherm1, 2019-01-31:
Per f2f discussions, with my own spin. Apparently users have two mostly-disjoint ways of thinking about multibody systems:
1. viewed as a collection of Model Instances, or
2. viewed as a single composite system, basically M v̇ = τ

Viewpoint 1 looks at quantities like force and velocities grouped conceptually by Model Instance, while viewpoint 2 sees only the underlying physics. There is no simple relationship between Model Instance groupings and the underlying physical (and computational) representation, which are chosen for efficiency and numerical accuracy, and subject to change with internal algorithm refinement. MBPlant provides utilities for mapping between the representations.

I think this makes good sense and we should make it easy to use either viewpoint. The problem @rcory found that @edrumwri reported above is due to incomplete support for these viewpoints. The inverse dynamics controller only supports viewpoint 2 (unified physical system), while MultibodyPlant supports only viewpoint 1 (per-model instance) actuation ports.

We can fix the immediate problem by teaching inverse dynamics about Model Instances and associated actuator orderings, or by teaching MultibodyPlant to accept full τ on some input port.

**Drake users**: please let us know your thoughts on this.

amcastro-tri, 2019-01-31:
Thanks @rcory, @edrumwri and @sherm1 for working on this. I think supporting the two viewpoints would be nice and espcially since we already had a couple of stories in which people expect 2, at least for simple fully actuated models.

For the short term, I think it does make sense to teach the IDC to understand model instances as @edrumwri did. In my mind that does seem like the right abstraction and it was only a bug that we didn't do it like that in the first place.

Supporting viewpoint 1 might be doable once #10068 merges in and we review the internal MBP tree structure. As @sherm1 pointed out, even if we internally shuffle dofs user code written in terms of MBP APIs should not be affected.

rcory, 2019-01-31:
As a side note related to Model Instances, I approached the implementation from viewpoint 2, where I was parsing the same `sdf` multiple times but expected the result to be a single composite model. The only reason I ended up with multiple model instances is because `Parser::AddModelFromFile` forces the user to create a new model instance for each parse, rather than providing the option of adding to an existing model instance, or preferably, in this case, allowing me to ignore model instances altogether. 

> We can fix the immediate problem by teaching inverse dynamics about Model Instances and associated actuator orderings, or by teaching MultibodyPlant to accept full τ on some input port.

@sherm1 I agree with the two options. For the first option, if we teach inverse dynamics controller about Model Instances and associated actuator ordering, then I think we'd also want inverse dynamics controller to provide multiple torque output ports (per model instance), each of which would directly connect to each corresponding MutlibodyPlant (model instance) actuation input port.

For the second option, if we teach MultibodyPlant to accept full τ on some input port, I think @jwnimmer-tri's slack comment makes sense, i.e., have the full τ be ordered by velocity index coming out of the inverse dynamics controller, and let MultibodyPlant's input port re-arrange it as it sees fit for internal use. 

The one remaining question in my mind is, what is the expected ordering for the _reference state_ input port of inverse dynamics controller (which isn't currently documented)? Perhaps it is up to the user to call `SetPositionFromArray` and `SetVelocitiesFromArray` when creating this reference state input, which I feel should at least be included in the inverse dynamics controller documentation for reference.

amcastro-tri, 2019-01-31:
> Perhaps it is up to the user to call SetPositionFromArray and SetVelocitiesFromArray when creating this reference state input, which I feel should at least be included in the inverse dynamics controller documentation for reference.

I think that's a great idea.

edrumwri, 2019-01-31:
> @sherm1 I agree with the two options. For the first option, if we teach inverse dynamics controller about Model Instances and associated actuator ordering, then I think we'd also want inverse [...]

This is fine, but since we're revisiting this architecture, we should strive for consistency. MBP exports all continuous state from one port. So it would seem that continuous state should then be output per model instance as well.

sherm1, 2019-01-31:
Per f2fs, I believe we have converged for now on just adding a τ (generalized force) input port to MBP ordered identically to generalized velocities. The particular ordering is not to be specified, just that the ordering is consistent (with mass matrix, Jacobians, etc.). The API already provides nice access routines for finding and setting particular entries.

No change is required to the ID Controller with this scheme. The Model Instance-oriented actuator ports to MBP remain and continue to function; anything fed into the τ port is additive.

tri-ltyyu, 2019-02-01:
This will be obsolete once the other feature is done.

sherm1, 2020-05-12:
@rcory @amcastro-tri can we close this now?

rcory, 2020-05-14:
There are a couple things that I think should be addressed before closing this issue. 

Inverse Dynamics documentation:
> The one remaining question in my mind is, what is the expected ordering for the reference state input port of inverse dynamics controller (which isn't currently documented)? Perhaps it is up to the [...]

Additionally, InverseDynamicsController states that its output is "torque", while `MultibodyPlant::get_generalized_force_input_port()` talks about "tau". I think for consistency the notation on both should be the same (maybe reference [System Dynamics](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#mbp_equations_of_motion))? Also, might be a good idea to call out in the documentation for InverseDynamicsController that its output port is to be connected to `MultibodyPlant::get_generalized_force_input_port()` **and not** `MultibodyPlant::get_actuation_input_port`, since for fully actuated systems these BasicVector ports would be exactly the same size and easily swapped by mistake (I've done this myself, and have seen others do it as well).

sammy-tri, 2020-05-14:
Did you mean `MultibodyPlant::get_applied_generalized_force_input_port`?

I started to write something longer a number of times and then (I hope) realized why I was getting confused...   which is that the uses of `InverseDynamicsController` I typically always use `MultibodyPlant::get_actuation_input_port`, since typically the controller is only generating torques for one model instance in a different, larger plant, and I don't think that's relevant to the use case being discussed here.

In terms of documentation changes, we should probably try not to confuse users of either configuration.

rcory, 2020-05-14:
@sammy-tri Yes, I did mean `MultibodyPlant::get_applied_generalized_force_input_port` (I've updated my above comment too).

Let me attempt to clarify what I understand to be the problem, and perhaps we can converge on what changes in documentation would be warranted. The system dynamics are given as:
`M(q)v̇ + C(q, v)v = τ = Bu`, where `τ` represents the generalized forces vector, `B` the actuator selector matrix, and `u` the actuation vector. The output of `InverseDynamicsController` is `τ`. My understanding is that `τ` is ordered according to velocity index ordering, and `u` is ordered according to actuator ordering. For `InverseDynamicsController` which only supports fully actuated systems we have `τ.size() == u.size()`. Therefore we have two options: 1) We convert generalized forces `τ` to actuation values `u` via the actuator selector matrix `B`, where the relationship `u = B.Inverse() * τ` holds. Then `u` can then be fed directly into `MultibodyPlant::get_acutation_input_port()`. Or, option 2) we directly feed `τ` (the output of IDC) into `MultibodyPlant::get_applied_generalized_force_input_port`. 

The fact that you are feeding the output of IDC (`τ`) into `MultibodyPlant::get_actuation_input_port()`  (`u`) and it still works seems to be lucky, because you have a single model instance in your control plant? If you were to have two model instances in your control plant, I believe the ordering `τ` and `u` would differ, and this wiring wouldn't work anymore. At least., that's what I've seen.

jwnimmer-tri, 2024-12-31:
I wonder if #22315 fully resolved this issue?