RobotLocomotion/drake #10547: leaf_system: Consolidate double overloads of `Derived::(*method)` + `std::function<>`

EricCousineau-TRI, 2019-01-31:
(Follow-up of Slack convo w/ @edrumwri)

When I've done the bindings of systems stuff, it was sometimes confusing when there was a mix of method pointer + `std::function<>` overloads for things like `DeclareX`.

Would be nice to clean this up, and consolidate to a single point that does the binding.

Example:
https://github.com/EricCousineau-TRI/repro/blob/3d79cc0/cpp/function_or_method.cc#L106-L126

EricCousineau-TRI, 2019-05-07:
Er... In doing the prototype and looking at the current use cases, it looks like it actually won't accommodate all cases of inference.

Here's the start, but I didn't even apply it to anything once I realized that inference suffers:
https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:feature/function_or_method-wip

For anything that infers parameters from callback arguments (e.g. ValueType for cache entries) can't work because it has too much indirection...

So that being said, this can be applied to a small subset of methods, but the real value gain only comes from these potential solutions:

1. Do nothing, and keep as is. (But mebbe still deprecate class method helpers? :D)
  a. Another alternative is to offer a generic binding mechanism, but still require users to step through that. I'd be a fan of this. Something like `DeclareOutputPort(Bind(&MyVectorCalc))`, and use `std::function<>` to infer argument types - however, lambdas still suffer...
2. Defer inference to after the function prototype resolves, like `pybind11` stuff:
  b. e.g. `DeclareStuff(auto my_callback) { using VectorType = decltype(my_callback)::arguments::type_at<3>; }`
3. Get rid of inference entirely, and always require user to specify the relevant parts; e.g. `DeclareAbstractState<FramePoseVector>(...)`.

I think all of these work involve more than I had envisioned...

jwnimmer-tri, 2021-11-12:
As I understand it, the new ValueProducer type has resolved this problem.

closed 2021-11-12 by jwnimmer-tri