RobotLocomotion/drake #10548: should std::uniform_real_distribution<symbolic::Expression> and friends support operator() with no arguments?

RussTedrake, 2019-01-31:
When showing this code:
https://github.com/RobotLocomotion/drake/blob/master/examples/manipulation_station/manipulation_station.cc#L487

I found myself explaining that the RandomGenerator instance here does absolutely nothing (ever); it's only there for compatibility with the std::uniform_real_distribution<double>::operator() syntax.  But we could easily support operator() with no arguments, which would clean up this code.  Is there any reason not to?

soonho-tri, 2019-01-31:
I'll add `operator()` with no arguments. I will justify it in its doc.

soonho-tri, 2019-01-31:
@RussTedrake , in #10549, you commented:

>  (though would be great to add the python binding, too?)

FYI, we do not expose `std::*_distribution`s to pydrake. We only provide 1) random symbolic variables (such as `RANDOM_UNIFORM` and `RANDOM_GAUSSIAN`), 2) random generators, and 3) `Expression::Evaluate(Generator)`.

RussTedrake, 2019-01-31:
ah.  then no action requested for now.  (will be playing with random symbolic expressions in python soon enough and will have better understanding then)

soonho-tri, 2019-01-31:
SGTM. Please open an issue if you have a feature-request!

closed 2019-01-31 by soonho-tri