RobotLocomotion/drake #10550: /opt is read-only on mac

RussTedrake, 2019-01-31:
When I run the instructions here:
https://drake.mit.edu/python_bindings.html#binary-installation-for-python
on my mac, it fails to create /opt/drake and its subdirectories.  Running as root was the easy fix, but it would be good to acknowledge that (and recommend this, or a better alternative) right in the doc.

RussTedrake, 2019-01-31:
I also think that https://drake.mit.edu/python_bindings.html#building-the-python-bindings should come later in the document.  Probably just before https://drake.mit.edu/python_bindings.html#debugging-with-the-python-bindings

cc @EricCousineau-TRI

jwnimmer-tri, 2019-01-31:
I think `/opt` in those docs was inspired by https://github.com/RobotLocomotion/drake-shambhala/blob/master/drake_cmake_installed/README.md (which does have the `sudo`).  However, I don't know _why_ we're using `/opt` -- that seems dumb.  Probably our instructions should always just use `$HOME`?

jamiesnape, 2019-01-31:
> However, I don't know why we're using /opt -- that seems dumb. Probably our instructions should always just use $HOME?

RPATH issues. Not dumb at all.

jwnimmer-tri, 2019-01-31:
A quick skim of an Ubuntu binary image shows that rpath and runpaths are all `$ORIGIN`-relative (or system `/usr/libx86_64/.```) stuff.  Is there an open issue, that we should have Drake's binary release be usable from any directory you unpack it into?  If not, we should create one.

jamiesnape, 2019-01-31:
This is configuration: mac.

jwnimmer-tri, 2019-01-31:
Fair enough.  But then we have another issue that the _Ubuntu instructions_ also say to use `/opt`, without sudo (so it _also_ doesn't work), and in any case on Ubuntu we should _not_ recommend `/opt`.

jamiesnape, 2019-01-31:
Our usage matches https://www.tldp.org/LDP/Linux-Filesystem-Hierarchy/html/opt.html in my opinion. People can install why they like however, but recommending `/opt` is not unusual.

jwnimmer-tri, 2019-01-31:
I agree that using `/opt/drake` is consistent with FHS (and Debian's additions to FHS).  My position is that it's bad UX for us to suggest doing so.  Putting files in `$HOME` is going to be a substantially better workflow, for Ubuntu users at least.  (Or for python, virtualenv + pip is actually what we should be shooting for once we have time.)

In any case, I'm okay with deferring that debate to a in a different issue / future work.  But the fact remains that for `/opt`, the read-only problem also exists on Ubuntu, so this "read-only" bug isn't exclusively `configuration: mac`.

jwnimmer-tri, 2019-03-20:
\CC @tri-ltyyu This is a documentation issue we hit last week.

jwnimmer-tri, 2021-11-02:
As best I can tell, as of #15953 none of our https://drake.mit.edu documentation tells the user to do things into `/opt` anymore.

The only place that does suggest `/opt` is https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed#instructions, so I guess fixing that documentation as well would be the last part of this issue.

RussTedrake, 2021-11-02:
Perhaps I just need a clarification, but I see a lot of `/opt` in e.g. https://drake.mit.edu/apt.html#stable-releases

jwnimmer-tri, 2021-11-02:
Yes, that's where our `apt` packages put Drake.  However, since that's `apt` (or really, `dpkg`) placing the files there, it works as intended, without the user need any special permissions or instructions, beyond what any other use of `apt install` would demand.

jwnimmer-tri, 2023-01-27:
This one is pretty easy.  The call to action is to change these instructions ...

https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed#instructions

... to unpack somewhere in the user's home directory, rather than `/opt`.  We should also run the new instructions by hand to make sure they work OK.

closed 2023-02-07 by BetsyMcPhail